omersiar/esp-rfid #239: unable to change reader type

trudeau102000, 2019-02-02:
on the settings- Hardware Settings the reader type-Pin D0-Pin D1 and Relay Pin cannot be changed
[image]

trudeau102000, 2019-02-02:
I have tried it on chrome, firefox, IE and android same result

omersiar, 2019-02-02:
When you flash forv2board.bin this is the expected behaviour.

If you do not have official board please use generic.bin

trudeau102000, 2019-02-02:
when ever i try to flash generic.bin instantly it pops up flash complete but nothing flashes to the module. any ideas?

trudeau102000, 2019-02-02:
the esp shows up on com3

trudeau102000, 2019-02-02:
i am suspecting either the esp or the cable. new ones on the way

omersiar, 2019-02-03:
it's same beside you do not need those settings on official hardware

trudeau102000, 2019-02-04:
guess I will just throw in the towel. i got a couple more esp's and some new cables and just ended up with the new issues. after uninstalling and reinstalling the driver i can now flash the firmware but after flashing it to probably 6 different ones its just one hurtle after another. now i can get the esprfid ap and connect but it don't matter what esp i do once i enter the wifi ssid a pass- save and commit using dhcp thats it. i can never connect to it again. no ap to connect to again and never connects to my wifi. after several days on this i'm done. wish i could remember what firmware i flashed the very first time because that one seemed to work and was pretty stable.

trudeau102000, 2019-02-04:
i now have it connected to my network and seems stable. i will be getting a couple new rc 522's tomarrow so i can continue. the only issue i have now is getting my relay functional i have it connected to gpio 14 but unable to get ti trigger with the test button. i can  trigger the relay manually by pulling the relay pin low or high

trudeau102000, 2019-02-04:
[image]
prob could have just went with the official hardware in the first place.and i will when i know this is going to work

omersiar, 2019-02-05:
Can you please download fresh 1.0.2 and delete forv2board.bin then rename generic.bin to forv2board.bin and then use flash.bat then chose option 2

trudeau102000, 2019-02-05:
I now have this working but the issue seems to be with the gain for therc522. if i select max 48db in hardware settings what should it show in the config for "rfidgain"?

trudeau102000, 2019-02-05:
[image]
[image]

trudeau102000, 2019-02-06:
> Can you please download fresh 1.0.2 and delete forv2board.bin then rename generic.bin to forv2board.bin and then use flash.bat then chose option 2

 I went with option 3 then generic.bin that got me back up and running. right now it will work but i have to keep changing the gain save and reboot to get it to read again after a short time. if you want me to try that i will.

trudeau102000, 2019-02-08:
[image]

nardev, 2019-02-09:
@trudeau102000 unfortunately any other reader but wiegand are quite demanding and with so much settings that it's sometimes hard to help.

please join http://esp-rfid.slack.com for chat help if you want.

Sometimes it's not a project issue, it's sometimes individual setup.

trudeau102000, 2019-02-10:
> @trudeau102000 unfortunately any other reader but wiegand are quite demanding and with so much settings that it's sometimes hard to help.
> 
> please join http://esp-rfid.slack.com for chat help if you want.
> 
> Sometimes it's not a project issue, it's sometimes individual setup.

thanks. nardev, i have several esp's and  readers. so i have been able to duplicate the same issues over several setups. i also have a simple uno setup to verify the readers. they will work perfectly on the uno  but stop functioning once moved to any of the esp's at one point i had the reader working but it was always short term. sometimes changing the rfgain and restarting would bring function back. but again only short term. I have what i need now via a mega 2560 I really liked this project but the results are disappointing.

msanchezt, 2019-03-03:
> @trudeau102000 unfortunately any other reader but wiegand are quite demanding and with so much settings that it's sometimes hard to help.
> 
> please join http://esp-rfid.slack.com for chat help if you want.
> 
> Sometimes it's not a project issue, it's sometimes individual setup.

How can I join this slack channel? Is asking for a specific domain which I do not own. Can't join with my own email address?

nardev, 2019-03-03:
@msanchezt can you give me your email? You can send it to esprfid@gmail.com too.

msanchezt, 2019-03-03:
Email sent, thanks.

*Marcel Sánchez Toledano*

Missatge de nardev <notifications@github.com> del dia dg., 3 de març 2019 a
les 20:22:

closed 2019-04-05 by omersiar