omersiar/esp-rfid #240: Backup User Data only returns some users?

nardev, 2019-02-14:
I just want to see of somebody can test this too.

I have issue when i want to Backup User Data, despite the fact that there is about 110 users in users list, if i try to backup that, i get only 7 in Backup file.

Can somebody confirm/check/test that?

omersiar, 2019-02-14:
Is there any information available browser's console? especially on Network tab?

nardev, 2019-02-14:
@omersiar i think that i figured out what is happening. Could you try this:

1. Login to web
2. Go to Users page and go thought the list a bit, left/right
3. Go to Backup page and press the Backup User Data

Only in that case this bug happens

Otherways, if you:

1. Login to web
2. Go straight to Backup page and press the Backup User Data

So, it's obviously JS bug, it should "reset" the page variable when it asks the device for users list or, there should be another function for extracting users, not page by page.

@omersiar was there any issues if you try to "get" all users at once instead of page by page? Any memory issues?

nardev, 2019-02-14:
Yea, here we should fix the "page" variable... It must start from the beginning.

```

function backupuser() {
  backupstarted = true;
  data = [];
  var commandtosend = {};
  commandtosend.command = "userlist";
  commandtosend.page = page;
  websock.send(JSON.stringify(commandtosend));
}
```

nardev, 2019-02-15:
@omersiar The page = 1; was forgotten in esprfid.js

```
function backupuser() {
  page = 1;
  backupstarted = true;
  data = [];
  var commandtosend = {};
  commandtosend.command = "userlist";
  commandtosend.page = page;
  websock.send(JSON.stringify(commandtosend));
}

```

closed 2019-02-15 by nardev