omersiar/esp-rfid #245: relay function?

trudeau102000, 2019-02-24:
I am wondering if anyone has been able to get a relay to function using nodemcu with Generic.bin? I finally have this working using nodemcu on development board (the breadboard setup  is just to buggy)  but can't get anything from my relay. its a 5v relay powered from a completely different source. i have gone through the setting options and probed all the gpio pins with a meter but see no change in state on any of them my relay will trigger on a low or high state. I added a button so i could test the relay and it works fine. i did find that if i plug power into the development board i will get  a constant low state from what ever gpio pin i select in settings. the second i click save and reboot the pin will go low triggering the relay then stay that way. this stops with the nodemcu powered by usb but i get nothing from the pins now.I would love to hear any suggestions and I am willing to load and test any firmware. we have been working with this project for a couple weeks now and have so far been able to overcome any issues but this one has me stumped. if i move everything over to a uno or a mega 2560 it works fine. i was hoping to get something from the relay gpio pin. even it it was a small change in state I could build off that to get the relay working. I assume even with the official hardware the relay pin is either pulled low or high  depending on relay type. we are trying to make a small wooden box that locks when closed but pops open when you scan a valid rfid card 12v lock solinoid off a 5v relay. i'm open to any suggestion from anyone. Thanks 
[image]

omersiar, 2019-02-25:
Can you please test with a previous version too? Some GPIOs of ESP8266 have weird behaviors, you need to select accordingly, also you can not toggle 5v relay with 3.3v logic, this is how Uno differs from ESP8266. You should probably failing this.

https://electronics.stackexchange.com/questions/296330/how-to-drive-a-relay-from-esp8266-gpio

trudeau102000, 2019-02-25:
thanks for the comment omersiar, after looking over my setup I realized that I was using a V3 nodemcu. no matter as to your point. but like i said I am powering the relay from a breadboard power supply so I have both 5v and 3.3v handy the relay will trigger fine from either the 3.3v  or the ground. like I said if there was even 3v from the relay pin I would have certainly used that to get what ever voltage I needed. if the pin would just pull low then problem solved. I don't know how the NodeMCU Lua V3 differs. would love to hear any input you have on that though. my next attempt is going to be with  a uno built for arducam. it has an embeded esp 8266. its built for a cam but i'm a glutten for punishment.lol. thanks again.
[image]

trudeau102000, 2019-02-25:
[image]
but hey i'm a sport

trudeau102000, 2019-02-27:
well here is an update. flashed generic.bin to this uno esp8266 board and it is functional, seems pretty stable so far web gui is solid rc522 is reading the cards( i have had to change the gain once so far to get it back)               however still unable to get any relay function. looks like the same issue. no change in relay pin state

[image]

trudeau102000, 2019-02-27:
if someone wants to put up a video of this as a functional rfid lock setup(official hardware or other) I would love to see it because from what I see it don't exist.

trudeau102000, 2019-02-28:
I didn't think so

Pako2, 2019-03-02:
Your image, which shows the interconnection of all necessary modules, there is little clear.
However - it seems that there is no connection between the GND of the ESP module and GND of the relay module. If I am not mistaken, then it is perfectly obvious why it does not work.

trudeau102000, 2019-03-02:
> Your picture (which shows the interconnection of all necessary modules) is rather unclear.
> However - it seems that there is no connection between the GND of the ESP module and GND of the relay module. If I am not mistaken, then it is perfectly obvious why it does not work.

true there is no ground from the esp to the relay Pako2. well not in that pic anyway. but i have been down that road over and over. setups have included the relay both powered and grounded from the esp 5v and ground i tried using the esp 3.3v through an l7805 with the esp ground.same results. however you are correct is saying it is perfectly obvious why it does not work. well to me it is anyway.lol

i tried using the esp 3.3v through an l7805

trudeau102000, 2019-03-02:
i have one last attempt planned i will try an OPTOCOUPLER with both 5v and ground on the relay signal side. but with no measureable change in the relay gpio pin i guess i will end up stuck with no way to switch the OPTOCOUPLER. which leads right back to where i am now

trudeau102000, 2019-03-02:
I thought i would get ahead of myself  so i cleared the flash and loaded fresh generic.bin but was unable to get esp-rfid connected to my network. finally after a few repeats of that i was able to get it connected only to be left with no reader function. i found a  RDM6300 in by  collection with some 125Khz cards so I gave that a shot only to end up with no save and reboot function from esp rfid.lol

trudeau102000, 2019-03-02:
i guess persistence is the key. one of my setups have function. not sure what changed but we were just trying to get the RDM6300 reading but were unable. went back to the rc522 and after about 10 save and reboots to get the settings to save its now reading and will turn an led on  off the relay pin. still on the uno esp board.will see what happens

trudeau102000, 2019-03-03:
been several hours and still stable. gonna be a few days to get an OPTOCOUPLER but with a signal on the relay pin now i should be able to drive it now with the led circuit and run 5v to the relay to trigger it. man I can't wait to wrap this up

Pako2, 2019-03-03:
I still do not understand your troubles. That's a trivial matter. It looks like you do not believe that this design works and you feel that there is a mistake in the project. There is no mistake (in this regard). If it is properly interconnected, it works.
**What version of relay module do you use?** Probably this is low level triggered version. This is completely inappropriate for this purpose, and it brings problems. You need a high lewel triggered version (for example [1-Channel 5V Relay Module High Level Trigger Expansion Board for Relays](https://www.aliexpress.com/item/1-Channel-5V-Relay-Module-High-Level-Trigger-Expansion-Board-for-Relays/32801470701.html)).
Then you will not need an optocoupler either. This would only be necessary if it is not possible to connect the GND of both power sources (5V and 3.3V). Of course, when you use an optocoupler, you can also use the low level triggered relay module too. But it does unnecessarily complicated wiring.

trudeau102000, 2019-03-03:
thanks Pako2 for the input, my problem outside of the completely random setting saving or the loss or user logs has for the most part been the relay issue. I trust you would agree that if there is no state change in the gpio pin for the relay its not really going to matter what relay is used. I have  stuck with this project  and put in a lot of time trying to get it to work for my needs. yes I have a low level relay but i am also running everything off the  uno with an embeded 8266. so the 5v, 3.3 and ground all share the same source.however the relay will trigger using either high or low to the signal pin.  currently I have this working with the 8266 uno board and rc522. if need be i will use a couple resistors and a transistor to convert the relay to a high level triggered version. and use the OPTOCOUPLER to drive it. as i said I now have a signal on the relay pin that will turn on an led when an authorized card is read so  this changes everything.thanks again

trudeau102000, 2019-03-05:
well to day i got my OPTOCOUPLER's today over the days waiting i have turned the board off and in the morning turned it back on and for a few days it was fine. plugged it in today and now no user data, won't read cards. wtf come on man. is this a joke or what?

trudeau102000, 2019-03-06:
throwing in the towel of this fucking garbage. which is what I should have done 3 weeks ago

omersiar, 2019-04-05:
Closing it for now, you may (and probably should) use the esp8266 forums.

closed 2019-04-05 by omersiar