omersiar/esp-rfid #246: MQTT without username and without password error

Klagio, 2019-03-01:
In my mqtt log (mosquitto) I have
1551462292: New connection from 192.168.1.187 on port 1883.
1551462292: Socket error on client <unknown>, disconnecting.

In esp-rfid log I have

Disconnected from MQTT server MQTT_NOT_AUTHORIZED
Disconnected from MQTT server TCP_DISCONNECTED

  "mqtt": {
    "enabled": 1,
    "host": "192.168.1.12",
    "port": 1883,
    "topic": "wiegand",
    "user": "",
    "pswd": ""

shouldn't be
username
password

instead of
user
pswd

?

omersiar, 2019-03-01:
it does not matter, it's just a json object, are you sure that your mqtt broker does allow unauthenticated publish?

Klagio, 2019-03-01:
Unauthenticated meaning?
I connect with other devices without issues. All inside my LAN. 

Below the config 

  "logins": [
    {
      "username": "",
      "password": ""
    }
  ],
  "anonymous": true,
  "customize": {
    "active": false,
    "folder": "mosquitto"
  },
  "certfile": "fullchain.pem",
  "keyfile": "privkey.pem"
}

omersiar, 2019-03-01:
esp-rfid currently do not support ssl connections

Klagio, 2019-03-01:
It's inside my LAN, no ssl

omersiar, 2019-03-01:
in configuration file it is defined so i assumed you are using it, i am no expert on mosquitto so i may be wrong.

Klagio, 2019-03-02:
No, I am not using SSL, so I think its not that

omersiar, 2019-03-02:
Now bug makes more sense, i need to investigate. Thank you for finding.

closed 2022-02-27 by omersiar