omersiar/esp-rfid #248: Not working (?) with this RFID reader

Klagio, 2019-03-02:
Hi, I bought this RFID reader (you can search for F9501D in Aliexpress)

https://www.aliexpress.com/item/SIB-Waterproof-Metal-Case-Keypad-RFID-Proximity-Card-Standalone-Access-Control-2000-Users-for-Outdoor-Indoor/32673908668.html?spm=a2g0s.9042311.0.0.27424c4dHiFdia

Then I flashed the ESP-RFID software on a Wemo D1 mini:  the webui is open correclty and all works fine (expet mqtt but that's not the issue here).

I then connected the Wiegand D0 to GPIO1  and Wiegand D1 to GPIO5 of the Wemo D1 Mini.

When I operate the RFID reader, nothing is shown in the access log of the ESP-RFID software.

Am I missing something obvious, or?

omersiar, 2019-03-02:
Please flash debug firmware and look for clues. Try another wiegand software to test hardware.

Klagio, 2019-03-02:
> Please flash debug firmware and look for clues. Try another wiegand software to test hardware.

Ok, will do. I put as READER TYPE: WIEGAND, maybe I should try others? (Now I dismantled everything, will have to remount it for testing)

Klagio, 2019-03-02:
> Try another wiegand software to test hardware.

I am very new to wiegand, any software you recommend for testing?

omersiar, 2019-03-02:
> I am very new to wiegand, any software you recommend for testing?

https://github.com/monkeyboard/Wiegand-Protocol-Library-for-Arduino

Klagio, 2019-03-02:
Ohh maybe is the voltage of D0 and D1? Will check the specs

Klagio, 2019-03-02:
I think my RFID reader exit D0 and D1 at 5V, is this a problem for a Nodemcu/Wemo?

omersiar, 2019-04-05:
GPIOs are 5v tolerant on ESP8266 but not the VCC pin it must be 3.3V

omersiar, 2019-04-05:
Closing it for now, you should debug this yourself.

closed 2019-04-05 by omersiar