omersiar/esp-rfid #249: Recommend RFID reader?

msanchezt, 2019-03-03:
Hi all,

Can someone recommend which is the best/good RFID reader for this Github project? I have been using it with the cheap RC522, but is giving me some troubles.
What is the RFID most tested? Or known to be the best for this specific project?

Thanks a lot.

MrG1981, 2019-03-18:
Hi
I can't tell if this is the best reader
but it's working flawless for me for over 6 months now and it is not expensive

€ 3,37  6%OFF | Promotion Price Wiegand 26/34 Rfid Card Reader for Access Control System MF IC 13.56Mhz Card Reader Access Control Slave Reader
https://s.click.aliexpress.com/e/bpHRaDHX

Best Regards
Gregor

omersiar, 2019-04-05:
I will prepare a recommendation section on README.

closed 2019-04-05 by omersiar

Shanghaiman, 2020-02-08:
> Hi
> I can't tell if this is the best reader
> but it's working flawless for me for over 6 months now and it is not expensive
>
> € 3,37 6%OFF | Promotion Price Wiegand 26/34 Rfid Card Reader for Access [...]

I'm trying to use a wieagnd rfid IC 13.56 MHz with nodmcu amica board.
The readings of the card numbers are never same.
Can you share ur settings?
Tks