omersiar/esp-rfid #250: Disconnected from MQTT server

js903801, 2019-03-03:
Hello, my original ESP RFID Relay board sporadically loses the connection to the MQTT server and does not reconnect automatically. Only after a reset is the MQTT will connect again. I read that I should use the version 0.8, here is the same problem. Just like the version 1.2. What can I do?

omersiar, 2019-04-05:
Can you please share mqtt logs?

omersiar, 2019-04-05:
MQTT Code should be revised. #256

cyberhasse, 2019-05-18:
Changing line:142 in main.cpp to unsigned long interval = 55; //1800;
works!

Great work btw.
Regards.

omersiar, 2019-05-18:
Interesting, this shouldn't be problem unless there are some rate limits on the server side. Thank you for sharing

moneybag, 2019-06-13:
Can you create a new bin file,  i dont have  platform io

Marsx79, 2019-06-13:
> Changing line:142 in main.cpp to unsigned long interval = 55;
> works for me!
> 
> Great work btw.
> Regards.

How can I adjust that? Is an Upadate planned? I also have the problem.

cyberhasse, 2019-06-13:
Compiling the dev version seems to fix the issue!
I can provide a bin file when i get home.

Regards.

Marsx79, 2019-06-13:
> Compiling the dev version seems to fix the issue!
> I can provide a bin file when i get home.
> 
> Regards.

That would be great. THX

cyberhasse, 2019-06-13:
The bin file is here:
https://drive.google.com/file/d/1fJ6QaL5geuwFs9yN2COBUVS5Ai8Jo2dy/view?usp=sharing
Regards.

moneybag, 2019-06-13:
Thx, i flashed it successfully.
lets wait and see

LEJOUI, 2019-08-13:
This solution worked for me. In v1.3 this problem was still there and caused a lot of mqtt disconnections.

LEJOUI, 2019-08-13:
Seems I was too early. Still having disconnected from mqtt after a few hours.

cyberhasse, 2019-08-14:
I bought a new router and after that the problem is better.
Regards.

LEJOUI, 2019-08-14:
Yes, still no luck over here.. It would be nice that after a drop in wifi or tcp disconnected there would be a few attempts to try to connect to the mqtt server again, or a setting that when tcp disconnects a restart will takes place. For now my solution is to disable wifi after x minutes so it reconnects to my mqtt server again. Downside is that i've to scan the tag twice after a long time not in use.

drthanwho, 2019-09-29:
Also on v1.3.3 and I'm getting MQTT disconnected after a few hours. For now I've set it to reboot every 24 hours but still not a solution since it disconnects after 3 - 4 hours. 

Anyone worked any possible solution for this?

Hini2001, 2021-05-16:
Hello and good Day to you All.
I have the Same Problem. I had esp rfid (actual ver 1.3.1) on a Node mcu and use it to control several Things over mqtt and IOBROKER. But After different time the mqtt Connection is closed. In the log file i could read:

WARN	mqtt	Disconnected from MQTT server
Additional Data	TCP_DISCONNECTED

No wlan issues and no wlan disconnections. Just the mqtt Connection is closing. Every other mqtt Client is still connected with the mqtt IOBROKER Server.
After a reboot of the esp rfid it establish the Connection again for sometimes hours or sometimes 1 Day... 
So i hope you can help me fixing this issue.

closed 2022-02-27 by omersiar