omersiar/esp-rfid #251: Wifi connection after power lost

araziforum, 2019-03-04:
Hi. I have a question about wifi connection. 
After power lost, the esp8266 can't connect to WiFi because it opens faster than the wifi modem. That's why it opens with its own AP name. Is it possible to resolve this situation, or will it be a software update? it's like trying to reconnect for about 2 minutes after the power comes in.

trudeau102000, 2019-03-04:
I have an uninterruptible power supply on my network  hardware UPS as well as a battery backup on my esp. if there is a power loss they are unaware it even happened.

omersiar, 2019-04-05:
This bothers me too, and should be fixed as soon as possible. I can not make promises but it is now high priority task.

frenchie71, 2019-09-03:
This reduced Wifi Wait time from 20 seconds down to 2.5 seconds in my case:

```
diff --git a/src/wifi.esp b/src/wifi.esp
index 28ef702..065ca62 100644
--- a/src/wifi.esp
+++ b/src/wifi.esp
@@ -93,8 +93,9 @@ void ICACHE_FLASH_ATTR fallbacktoAPMode()
 // Try to connect Wi-Fi
 bool ICACHE_FLASH_ATTR connectSTA(const char *ssid, const char *password, byte bssid[6])
 {
+       WiFi.persistent(false);
        WiFi.mode(WIFI_STA);
-       WiFi.begin(ssid, password, 0, bssid);
+       WiFi.begin(ssid, password);
 #ifdef DEBUG
        Serial.print(F("[ INFO ] Trying to connect WiFi: "));
        Serial.print(ssid);

```
The reason might be that if you have multiple access points it always tries to connect to the same (maybe weaker) AP using the given BSSID. Disabling SDK Autoconnect and removing the BSSID solves the issue in this case as it just takes the strongest available AP

frenchie71, 2019-09-03:
@araziforum - if you go to "Setup" - "Wireless Network" in the web interface, is the BSSID field filled out, i.e. does it contain something like B0:A2:C3:35:67:98 (or similar)?

If it is empty, it might be worth clicking on the "scan" button and actually select the wifi from there so that the BSSID field gets filled because I have realized that the setup routine tries to convert the BSSID field even if it is empty, thus worst case filling the BSSID with garbage. The ESP then first tries to connect to a non-existent BSSID until it times out before it roams to another AP.

> Hi. I have a question about wifi connection.
> After power lost, the esp8266 can't connect to WiFi because it opens faster than the wifi modem. That's why it opens with its own AP name. Is it possible [...]

spatnynick, 2020-06-22:
Hi all, this is a problem for me too. As I am using rfid in long-term production use, sometimes it happens that Wifi is not available from any reason when esp-rfid starts. Then esp switches to AP mode forewer, it newer re-tries to connect to known/configured SSID, even though I have Auto-restart set to 24 hours (does it work in AP mode?). One has to go with a mobile to the are where AP-signal is reachable, connect, goto 192.168.4.1 and restart esp-rfid manually.

My suggestion is a feature like:
- AutoRestart when Wifimode = Client but Wifi is not connected (running in AP mode) every 30minutes; auto-switching to AP mode is fine&working please don't change that

Br Bohuslav

Empire99, 2020-06-30:
I also have this issue, after wifi gets back, The device never reconnects to wifi. It should try to reconnect in client mode from time to time.

omersiar, 2022-04-04:
Addressed with #495.

matjack1, 2022-04-04:
@araziforum @spatnynick @Empire99 @frenchie71 I've addressed this with this PR: https://github.com/esprfid/esp-rfid/pull/495 on the dev branch.

You can now disable the fallback AP mode and if you do that it will wait and try reconnecting to WiFi every minute.

I'm closing this for now, but reopen if necessary, thanks!

closed 2022-04-04 by matjack1

Empire99, 2022-04-05:
Hello, can someone add binary files with latest changes? I am trying to buid it with platform io and only get a lot of errors.

matjack1, 2022-04-05:
hey @Empire99 you can get the binaries from here: https://github.com/esprfid/esp-rfid/suites/5917945892/artifacts/202137764

While for the compilation, you should use the `dev` branch and make sure to run a clean first so that it gets the latest dependencies. Let me know how it goes! :)