omersiar/esp-rfid #255: Mqtt control relay switch

zhangziran, 2019-03-25:
Is there another way to implement Wiegand to read encrypted sectors?

zhangziran, 2019-03-26:
.

Netoperz, 2019-03-30:
wiegand, no. You need PN532 for example and some code to get the data, it's not a problem to do that , there are examples on the pn532 lib to red data, and on the gui you should add the place to input the password if you want to use own encrypted tada not the satandard keys.

As for wiegand, You need to buy reader that can work that way, for example ROGER has got this functions where you may preprogram the reader to read choosen sector with a default or predefined by admin key to decipher, and than it is throwing the output to the wiegand where you could catch the data using this device.

zhangziran, 2019-03-30:
> wiegand, no. You need PN532 for example and some code to get the data, it's not a problem to do that , there are examples on the pn532 lib to red data, and on the gui you should add the place to [...]

Hello, thank you very much for your reply and patient answers.

closed 2019-04-05 by omersiar