omersiar/esp-rfid #256: Problem mit Wlan/ Mqtt

DeUser, 2019-03-26:
Hallo zusammen, ich schreibe einfach mal auf deutsch.
Ich habe da ein Problem und zwar scheint es bei mir zu MQTT Verbindungsabbrüchen in Verbindung mit einem Wlan ausfall zu kommen.
Sporadisch meldet sich mein ESP Rfid nichtmehr auf meiem MQTT im Iobroker an. Wenn ich den ESP Rfid dann aber neu starte funktioniert es sofort wieder.
Wie kann ich ihn dazu zwingen sich automatisch wieder bei dem MQTT Broker anzumelden ohne jedesmal selbst hand anzulegen?

Netoperz, 2019-03-29:
english please

omersiar, 2019-04-05:
Es tut mir leid für dieses Problem. Die MQTT-Funktion wird überarbeitet und korrigiert. Sie müssen möglicherweise eine Weile warten.

omersiar, 2019-04-05:
#250

omersiar, 2019-04-05:
Let's track this on #250. Closing for now.

closed 2019-04-05 by omersiar