omersiar/esp-rfid #257: Use wserver.js

78Ronin78, 2019-03-28:
Hello! I have one question, can I use a web server emulator, vsverver, zhs as a server for writing data to the database SGLite for example? Or is it used only to configure the web interface and various tests? I do not ask to tell how to do it, I wonder if this is possible or not? Thank you in advance for the answer, I only study programming and many things are still not clear to me.

omersiar, 2019-03-28:
It just simulates websocket messages that are normally coming from the esp8266. I believe there are no other use case available for it.

78Ronin78, 2019-03-28:
Last question on this issue - can i configure this emulator websocket for data transfer? I making procedure that saving data on SQLite data base. It remains only to transmit this data to the ESP via use mqtt-protocol.

omersiar, 2019-03-28:
A nodejs application is totally capable of doing that.

78Ronin78, 2019-03-28:
Well, thank you so much! I have no more questions.

closed 2019-03-28 by 78Ronin78