omersiar/esp-rfid #259: What is the difference between manually adding a user and a tag?

78Ronin78, 2019-03-31:
I expanded your project to include support for the database SQLite3. This is the server part. That is, I am sending data to the database from the emulator of the websocket. But at the same time I enter the data on the  tag in manual. I do not use the reader. Question: if the data on the card is driven in after the contact of the reader and the tag, then why do we need to manually enter the data on the card? How will a new record be tied to the tag during manual input?

omersiar, 2019-03-31:
esp-rfid does not enter any data on any card. There is no difference between manually adding a user or by scanning a tag, if you know the UID of the card you can add it manually.

78Ronin78, 2019-04-01:
Thank much!

closed 2019-04-01 by 78Ronin78