omersiar/esp-rfid #260: Problem to save parameter

chryss44, 2019-04-05:
I have try 1.10.2 on Wemos D1 Mini and everything is perfect on boot, but when a change parameters (like wifi or other) save, and accept change, nothing save and the Wemos doesn't reboot.
I have tried Generic, V2 and debug, but same result.
On reboot, all parameters are default :-(.
A have tried on Wemos D1 Mini pro too and it's the same.

Do you have a solution ?

omersiar, 2019-04-05:
Which os/browser do you use? Do you think you can use browser's developer tools to debug this issue?

chryss44, 2019-04-05:
I have used Safari and Chrome on last version Mac Osx (10.14.2).
i ll try Chrome on Windows 7 this evening and can use developper tools to look at error on MAC, ok.

chryss44, 2019-04-05:
The console result on Chrome (OSX 10.14.2)
```
192.168.4.1/:233 GET http://192.168.4.1/fonts/glyphicons-halflings-regular.woff2 net::ERR_ABORTED 404 (Not Found)
(index):1 [DOM] Input elements should have autocomplete attributes (suggested: "new-password"): (More info: https://goo.gl/9p2vKq) <input id=​"password" class=​"form-control" placeholder=​"Password" name=​"password" type=​"password" value required title=​"Please enter your password">​
(index):1 [DOM] Password field is not contained in a form: (More info: https://goo.gl/9p2vKq) <input class=​"form-control input-sm" placeholder=​"Administrator Password" id=​"adminpwd" type=​"password">​
(index):1 [DOM] Password field is not contained in a form: (More info: https://goo.gl/9p2vKq) <input class=​"form-control input-sm" placeholder value style=​"display:​inline;​max-width:​185px" id=​"mqttpwd" type=​"password">​
(index):1 [DOM] Password field is not contained in a form: (More info: https://goo.gl/9p2vKq) <input id=​"wifipass" class=​"form-control input-sm" name=​"ap_passwd" type=​"password">​
(index):1 [DOM] Password field is not contained in a form: (More info: https://goo.gl/9p2vKq) <input id=​"wifipass" class=​"form-control input-sm" name=​"ap_passwd" type=​"password">​
```

omersiar, 2019-04-05:
go to network tab on dev console and select WS tab below and watch what happens when hitting save settings

chryss44, 2019-04-05:
The result is :
```
GET http://192.168.4.1/fonts/glyphicons-halflings-regular.woff2 net::ERR_ABORTED 404 (Not Found)
(index):1 [DOM] Input elements should have autocomplete attributes (suggested: "new-password"): (More info: https://goo.gl/9p2vKq) <input id=​"password" class=​"form-control" placeholder=​"Password" name=​"password" type=​"password" value required title=​"Please enter your password">​
(index):1 [DOM] Password field is not contained in a form: (More info: https://goo.gl/9p2vKq) <input class=​"form-control input-sm" placeholder=​"Administrator Password" id=​"adminpwd" type=​"password">​
(index):1 [DOM] Password field is not contained in a form: (More info: https://goo.gl/9p2vKq) <input class=​"form-control input-sm" placeholder value style=​"display:​inline;​max-width:​185px" id=​"mqttpwd" type=​"password">​
(index):1 [DOM] Password field is not contained in a form: (More info: https://goo.gl/9p2vKq) <input id=​"wifipass" class=​"form-control input-sm" name=​"ap_passwd" type=​"password">​
(index):1 [DOM] Password field is not contained in a form: (More info: https://goo.gl/9p2vKq) <input id=​"wifipass" class=​"form-control input-sm" name=​"ap_passwd" type=​"password">​
required.js:12 [Violation] 'setTimeout' handler took 56ms
[Violation] Forced reflow while executing JavaScript took 55ms
```

chryss44, 2019-04-05:
I have tried on Windows 7 with Chrome ans same problem.

chryss44, 2019-04-06:
I have tried jut user add and same issue :-(, i can't save configuration.

chryss44, 2019-04-14:
I have found the problem :-).
I have used Atom to program my Wemos D1 Mini and everything is perfect.
The problem was from esptools, i don't know why i used this command and doesn't work (impossible configuration save). 
`esptool.py --port /dev/cu.wchusbserial1410 write_flash -fm dio -fs 1MB 0x00000 generic_rfid.bin`

chryss44, 2019-04-16:
The problem is from my program (esptools), i think parameters are not correct for Wemos D1 mini. I'll try to change parameters and post there.

closed 2019-04-16 by chryss44