omersiar/esp-rfid #269: Using the Wemos D1 Mini, is it possible to incorporate a door state sensor?

arretx, 2019-05-28:
I've managed to enable and disable the lock via the reader, but there are two additional functions that I need to figure out.

1.  I need to be able to use a reed switch to sense the state of the door (open or closed)
2.  I need to be able to connect a switch to the microcontroller to manually disable the magnetic lock.

With the current firmware, is this a possibility?  Wemos D1 Mini with Wemos D1 Mini Relay and RFID reader.  One manual switch.  One reed switch.

Thanks!

wimpie007, 2019-05-31:
would like to see this option too!

andersonthomas, 2019-05-31:
that would be a great options !

zonk-242, 2019-06-06:
The software does have code for Door Open and Door Closed logging locally and via MQTT.
The official hardware board can use a door sensor on GPIO-16.
Through the Web interface you can configure via |Settings|Hardware Settings|Door Status|

arretx, 2019-06-06:
If I'm using a NodeMCU1.0, would the same ring true or is it just the official board?

zonk-242, 2019-06-13:
> If I'm using a NodeMCU1.0, would the same ring true or is it just the official board?

If the NodeMCU1.0 board has GPIO16 then yes it does ring true.

andersonthomas, 2019-06-24:
im sorry, but i cant see any option for door status at hardware options.

omersiar, 2019-08-26:
I'm closing this for now, please feel free to reopen if you want additional help.

closed 2019-08-26 by omersiar