omersiar/esp-rfid #272: Invalid head of packet (0xE0)

ccfiel, 2019-06-08:
Configuring flash size...
A fatal error occurred: Invalid head of packet (0xE0)

Any idea why I have this error when I execute this step `platformio run -e generic -t upload` ?

zonk-242, 2019-06-13:
Looks like platformIO cannot find your ESP device.
Did you check to see that the Wemos D1 is connected to Ubuntu? Which Com port is it using?

Also try to reset or power cycle the Wemos D1 just before you upload from platformIO. I've seen my ESP8266 fail to upload if it's been on for a while.

ccfiel, 2019-06-13:
@zonk-242 I think its not the Wemos because when I use directly esptool.py it works perfectly.
```
esptool.py --port /dev/ttyUSB0 write_flash -fs 4MB -fm dout 0x0 firmware.bin
```

fwestenberg, 2019-08-14:
I had the same issue when flashing my Nodemcu. Solved it by changing the upload_speed parameter in platformio.ini to 115200.

omersiar, 2019-08-26:
I'm closing this for now, please feel free to reopen if you want additional help.

closed 2019-08-26 by omersiar

jamesawebb, 2019-11-03:
+1 on changing baud from default of 921600 to 115200.