omersiar/esp-rfid #273: Is it possible to control the door lock remotely?

arretx, 2019-06-11:
I'm able to use the web interface to setup the device, and have been able to use the RFID tags to trigger the relay.

How do I remotely control the relay using MQTT?  Isn't there a way to publish a topic to force the relay to switch without needing to be at the device with a keyfob?

marelab, 2019-06-12:
I implement that feature in my fork of esp-RFID https://github.com/marelab/esp-rfid. Beside opening the door over MQTT I implented user managment functions & logging.

cyberhasse, 2019-06-13:
@marelab
Hi testing your version with great success..
Would you be so kind and implement the door sensor funktion also?
Tried it in dev version here and it was great :)

Regards.

marelab, 2019-06-13:
You ment to get the sensor events over MQTT? If so I will look the next
days after it at my fork, the next version of my fork will publlish all
events that are logged over MQTT and disable local device logging.
best regards
marc

omersiar, 2019-06-13:
@marelab If you willing to share your code with the esp-rfid community through mainstream repository, I can give you commit access. Thank you for your contributions.

omersiar, 2019-06-13:
I can also put bounties on this one, to compansate your efforts.

cyberhasse, 2019-06-13:
@marelab yes i mean the door sensor over mqtt and maybe changeable high/low sensing..
Love all of the work on both versions.
Regards.

marelab, 2019-06-13:
Oh sure omersiar but I added also a few GUI elements to enable MQTT logging nothing big mostly expanded the MQTT functions and a few changes having flags on the logging. Just finishing a hole Node-RED flow where you can manage several devices of a home installation over MQTT llike open door add / edit / delete user. The MQTT stuff I have documentate at my fork read.me at the top under  https://github.com/marelab/esp-rfid. Maybe also take a look on my hardware design that doesn't need a sperate power source under https://github.com/marelab/rfid-door drives the door coil over a Mosfet and a few other changes to your great design. Take a look and if you like lets get in contact.
Best regards 
marc

cyberhasse, 2019-06-14:
A Doorbell button wich reports to mqtt would be awesome too!
Sorry if this went out of topic.
Regards.

marelab, 2019-06-14:
@cyberhasse you can use the door status pin for that if you don't need it for its orignal purpose. The door status will be signaled over MQTT in the ESP-RFID dev version.

cyberhasse, 2019-06-14:
Yes i know im already playing with your dev branch and it is great..
Thank you all for your good work.
I actually use it for door open/closed status so the extra pin for a
Doorbell would be the finishing touch :)
I noticed that you will have to update the webgui with mqtt settings and doorpin if im not mistaking.
Coped over from omersair dev before i compiled but mqtt logging stuff was missing.
Regards.

omersiar, 2019-06-21:
Please feel free to test. @marelab thanks again for making it real thing.

https://github.com/esprfid/esp-rfid/releases/tag/1.3.1

omersiar, 2019-08-26:
I'm closing this for now, please feel free to reopen if you want additional help.

closed 2019-08-26 by omersiar