omersiar/esp-rfid #276: is there a way to reset all settings without flashing again

vetruk, 2019-06-17:


omersiar, 2019-06-17:
If you do not want retain user data as well, you can use Factory Reset.

omersiar, 2019-06-17:
Or you can backup user data and do the factory reset then restore them.

vetruk, 2019-06-17:
The problem that ocurs often is that when wifi password is set and if is
shorter than 8 symbols device become inresponsive. Sometimes even with more
than 8 symbols. Is there a way to soft  reset settings and data anyway.
What does mean Factory reset.

vetruk, 2019-06-17:
How about some dedicated button support for soft reset on WiFi settings.
Or all settings. On startup function to reinitialise setting file. Like if its pressed on startup to start to wipe the flash file system and making new files.

omersiar, 2019-06-17:
A typical WPA passphrase should be consist from at least 8 characters. I think code doing a check for that but I'm not sure at the moment. We have had the push button reset feature in the past, to reduce number of used GPIO I removed the feature. With the factory reset, I mean a menu item that is listed on the left menu, you need functioning WebUI to use it though.

vetruk, 2019-06-17:
I check the code and no lenth check so far. When wifi password is less than
8 characters the esp lock it self and constantly trying to wake up wireless
and freezes lime that and only solution that i found so far is to program
with empty bin and so on.. If you dont have problem to look at that code
that have in the past will be nice!

omersiar, 2019-08-26:
I'm closing this for now, please feel free to reopen if you want additional help.

closed 2019-08-26 by omersiar