omersiar/esp-rfid #277: ISR not in IRAM!

mandreko, 2019-06-19:
I'm trying to get esp-rfid setup on some nodemcu chips (bought here: https://amzn.to/2INQUCY) with a weigand reader (currently detached for troubleshooting). I saw that my chip is DIO instead of QIO, so I'm not sure if that's causing issues. I can upload the generic firmware from 1.0.2, and it will typically boot, but when I change the wifi settings to 'client', it no longer starts up. 

I setup platformio to do a custom build, since I saw https://github.com/esprfid/esp-rfid/issues/91 (although there is no flag in platformio.ini to change any longer). I ran the 'upload and monitor' task with the 'debug' build environment. When I do that, I see my config file that I setup my wireless information in it, but it gives an interesting error message that I haven't seen anywhere else, and apparently crashes.

Here's the log:
```
> Executing task: C:\Users\mandr\.platformio\penv\Scripts\platformio.exe run --target upload --target monitor --environment debug <

Processing debug (platform: https://github.com/platformio/platform-espressif8266.git; board: esp12e; framework: arduino)     
-----------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 160MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson> 5.13.4 
|-- <ESPAsyncTCP> 1.2.0  
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncUDP>        
|-- <ESP Async WebServer> 1.2.0
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ArduinoJson> 5.13.4
|-- <AsyncMqttClient> 0.8.2
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|-- <MFRC522> 1.4.1 #ea7ee3f
|   |-- <SPI> 1.0
|-- <Wiegand-Protocol-Library-for-Arduino> #1c2b580
|-- <Time> 1.5
|-- <Bounce2> 2.52
|-- <SPI> 1.0
|-- <Wire> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <Ticker> 1.0
Retrieving maximum program size .pioenvs\debug\firmware.elf
Checking size .pioenvs\debug\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  43.9% (used 35928 bytes from 81920 bytes)
PROGRAM: [=====     ]  53.0% (used 554028 bytes from 1044464 bytes)
Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: COM10
Uploading .pioenvs\debug\firmware.bin
esptool.py v2.6
Serial port COM10
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: cc:50:e3:3c:ab:ba
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 558176 bytes to 425260...
Wrote 558176 bytes (425260 compressed) at 0x00000000 in 5.5 seconds (effective 817.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
=============================================== [SUCCESS] Took 18.48 seconds ===============================================

--- Available ports:
---  1: COM1                 u'Communications Port (COM1)'
---  2: COM3                 u'USB5734 COM Port (COM3)'
---  3: COM8                 u'Standard Serial over Bluetooth link (COM8)'
---  4: COM9                 u'Standard Serial over Bluetooth link (COM9)'
---  5: COM10                u'Silicon Labs CP210x USB to UART Bridge (COM10)'
--- Enter port index or full name: 5
--- Miniterm on COM10  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

[ INFO ] ESP RFID v1.0.2
Flash real id:   00164068
Flash real size: 4194304

Flash ide  size: 4194304
Flash ide speed: 40000000
Flash ide mode:  DIO
Flash Chip configuration ok.

[ INFO ] Config file found
{
  "command": "configfile",
  "network": {
    "bssid": "78:8A:20:87:4B:DF",
    "ssid": "IoT",
    "wmode": 0,
    "hide": 0,
    "pswd": "redacted_pass",
    "offtime": 0,
    "dhcp": 1,
    "ip": "",
    "subnet": "",
    "gateway": "",
    "dns": "",
    "apip": "192.168.4.1",
    "apsubnet": "255.255.255.0"
  },
  "hardware": {
    "readerType": 1,
    "wgd0pin": 4,
    "wgd1pin": 5,
    "sspin": 0,
    "rfidgain": 32,
    "wifipin": 255,
    "rtype": 1,
    "ltype": 0,
    "rpin": 4,
    "rtime": 400,
    "buttonpin": 255
  },
  "general": {
    "hostnm": "esp-rfid",
    "restart": 0,
    "pswd": "admin"
  },
  "mqtt": {
    "enabled": 0,
    "host": "",
    "port": 1883,
    "topic": "",
    "user": "",
    "pswd": ""
  },
  "ntp": {
    "server": "pool.ntp.org",
    "interval": 30,
    "timezone": 0
  }
}
[ INFO ] Trying to setup RFID Hardware
ISR not in IRAM!

Abort called

>>>stack>>>

ctx: cont
sp: 3ffffc20 end: 3fffffc0 offset: 01b0
3ffffdd0:  3ffe9c4a 00000000 3ffe9c49 4021bc32  
3ffffde0:  3ffe9c4a 00000000 3ffe9c49 00000004  
3ffffdf0:  00000005 00000005 00000000 40100d72  
3ffffe00:  4025d9f2 00000065 3fff1424 40212159  
3ffffe10:  00000004 3ffe9491 3fff23b8 402027c5  
3ffffe20:  3ffe9c48 3fff2390 3fff2298 00000004  
3ffffe30:  3fff04b8 3fff2390 3fff2298 40244b30  
3ffffe40:  3ffe9675 3fff2390 3fff2298 40246144  
3ffffe50:  40222f20 00000000 000003e8 feefeffe  
3ffffe60:  3fff1eac 3fff1eec 00000000 00000000  
3ffffe70:  3fff0728 3ffffe7d 3ffffea4 00000000  
3ffffe80:  3fff0758 00000240 40222ad8 4021c304  
3ffffe90:  3fff2384 00000400 3ffffe74 0000036a  
3ffffea0:  00000000 3fff04b8 00000020 4021c548  
3ffffeb0:  00000000 3fff0394 00000010 4022b1ea  
3ffffec0:  00000000 3fff1e64 3fffff50 402069d1  
3ffffed0:  4021b664 3fff1e64 3fffff50 40206de0  
3ffffee0:  00000000 00000000 00000000 00000000  
3ffffef0:  3fff1f04 00001fb0 402069fc 402066dc  
3fffff00:  3fff24a0 3fff2518 3fff2458 00000000  
3fffff10:  00400000 3fff1208 3fff009c 402026dc  
3fffff20:  00000000 00000000 3fff04b8 3fff0728  
3fffff30:  3ffe9675 3ffe8dd4 3fff0390 40246857  
3fffff40:  402025a8 feefeffe 40221834 40221820  
3fffff50:  00000000 00000000 feefeffe feefeffe  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  feefeffe feefeffe feefeffe feefeffe  
3fffff80:  00000002 feefeffe feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3fff05ec  
3fffffa0:  3fffdad0 00000000 3fff05bc 40219d80  
3fffffb0:  feefeffe feefeffe 3ffe858c 40101045  
<<<stack<<<
��␁P�z%�
[ INFO ] ESP RFID v1.0.2
Flash real id:   00164068
Flash real size: 4194304

Flash ide  size: 4194304
Flash ide speed: 40000000
Flash ide mode:  DIO
Flash Chip configuration ok.

[ INFO ] Config file found
```

It then just keeps cycling through reading my config file, setting up the RFID hardware, and then giving the isr and iram error message with a stack trace.

ver2go, 2019-06-19:
Comment and replace platform.ini

< ;platform = ${common.platform}
< platform = espressif8266@2.0.4

mandreko, 2019-06-19:
Thanks, that at least got me to where I can debug stuff. 

Is there some breaking change in newer versions past 2.0.4 which cause issues?

ver2go, 2019-06-19:
I'm not a developer but play one on the internet :).

About a month ago I saw this issue and from googling, tracked it down to

6092 5995 I think

https://github.com/esp8266/Arduino/releases/tag/2.5.1

closed 2019-06-19 by mandreko

beikeland, 2019-07-17:
> Comment and replace platform.ini ;platform = ${common.platform} platform = espressif8266@2.0.4

I think this should fix the underlying library to run under the latest esp8266 code
https://github.com/monkeyboard/Wiegand-Protocol-Library-for-Arduino/pull/38

ccfiel, 2019-08-09:
@beikeland  I tried to use your repo still got the error. Do I need to do something else?