omersiar/esp-rfid #278: v1.3.1 Test Release stuck after saving settings

teriall, 2019-06-22:
[ INFO ] ESP RFID v1.3.1
Flash real id:   00164068
Flash real size: 4194304

Flash ide  size: 4194304
Flash ide speed: 40000000
Flash ide mode:  DIO
Flash Chip configuration ok.

[ INFO ] Config file found
{
  "command": "configfile",
  "network": {
    "bssid": "",
    "ssid": "esp-rfid",
    "wmode": 1,
    "hide": 0,
    "pswd": "",
    "offtime": 0,
    "dhcp": 1,
    "ip": "",
    "subnet": "",
    "gateway": "",
    "dns": "",
    "apip": "192.168.4.1",
    "apsubnet": "255.255.255.0"
  },
  "hardware": {
    "readertype": 1,
    "wgd0pin": 12,
    "wgd1pin": 14,
    "sspin": 0,
    "rfidgain": 32,
    "wifipin": 255,
    "rtype": 1,
    "ltype": 0,
    "rpin": 4,
    "rtime": 400,
    "buttonpin": 255,
    "doorstatpin": null,
    "openlockpin": null
  },
  "general": {
    "hostnm": "esp-rfid",
    "restart": 0,
    "pswd": "admin"
  },
  "mqtt": {
    "enabled": 0,
    "host": "",
    "port": 1883,
    "topic": "",
    "user": "",
    "pswd": "",
    "syncrate": 180,
    "mqttlog": 0
  },
  "ntp": {
    "server": "pool.ntp.org",
    "interval": 30,
    "timezone": 0
  }
}
[ INFO ] Trying to setup RFID Hardware
ISR not in IRAM!

Abort called

>>>stack>>>

ctx: cont
sp: 3ffffc10 end: 3fffffc0 offset: 01b0
3ffffdc0:  3ffe9e6d 00000000 3ffe9e6c 4021c5ce  
3ffffdd0:  00000000 4bc6a7f0 00000485 0000000c  
3ffffde0:  0000000e 0000000e 00000000 40100d92  
3ffffdf0:  4025e6f2 00000065 40100a86 40212ad1  
3ffffe00:  0000000c 3ffe96ac 3fff27a0 40202889  
3ffffe10:  3fff28a8 3fff2680 3fff2778 0000000c  
3ffffe20:  3fff28a8 3fff2680 3fff2778 40245568  
3ffffe30:  3ffe989b 3fff2680 3fff2778 40246df0  
3ffffe40:  40223958 00000000 000003e8 3ffffe50  
3ffffe50:  3fff224c 3fff228c 00000000 00000000  
3ffffe60:  3fff0950 402238d8 00000000 0000000a  
3ffffe70:  3fffff20 3ffffef0 40223500 00000020  
3ffffe80:  3fff297c 00000800 3ffffe64 000003af  
3ffffe90:  00000000 3fff06e0 3fff2220 3fff21e4  
3ffffea0:  00000000 4022bc22 00000010 4021cee4  
3ffffeb0:  00000000 3fff05bc 00000010 4022bc22  
3ffffec0:  00000000 3fff2204 3fffff50 40207069  
3ffffed0:  4021c000 3fff2204 3fffff50 40207478  
3ffffee0:  3fff22a4 00000000 00000000 00000000  
3ffffef0:  3fff2204 00001fb0 40207094 3fff2940  
3fffff00:  3fff2860 00000000 00000000 00000000  
3fffff10:  00400000 3fff15a8 3fff02c4 402027a0  
3fffff20:  00000000 00000000 3fff06e0 3fff0950  
3fffff30:  3ffe989b 3ffe8e18 3fff05b8 4024755f  
3fffff40:  40202604 feefeffe 40222238 40222224  
3fffff50:  00000000 00000000 feefeffe feefeffe  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  feefeffe feefeffe feefeffe feefeffe  
3fffff80:  00000002 feefeffe feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3fff0814  
3fffffa0:  3fffdad0 00000000 3fff07e4 4021a720  
3fffffb0:  feefeffe feefeffe 3ffe858c 40101065  
<<<stack<<<

ver2go, 2019-06-22:
https://github.com/esprfid/esp-rfid/issues/277

omersiar, 2019-06-22:
I will prepare an another release. Thanks for the heads up.

omersiar, 2019-08-26:
I'm closing this for now, please feel free to reopen if you want additional help.

closed 2019-08-26 by omersiar