omersiar/esp-rfid #279: v1.3.1 - no MQTT for RFID tags - only sync (heartbeats) and boot message

bib2, 2019-06-23:
Hi, i updated to 1.3.3.

But webinterface still shows 1.3.1

My problem now: I don´t get any messages when someone holds a rfid tag to my reader.

The reader is working, the led changes the color and i hear the beep, but no mqtt-messages anymore.

I get heartbeat mqtt messages every second, then i cahnged the new value for "sync interval" and now i get every 10 minutes - so mqtt is working for me. I get a mqtt message in boot too.

I wanted to change mqtt logging to enabled, but after a restart, it is disabled again.

I don´t use the internal user verification for my rfid tags, i send the whole tag information to my smarthome server and there i have my rules... So now when i don´t get any information over mqtt, i can´t access my house with rfid anymore.

omersiar, 2019-06-23:
I'm sorry to hear that, you can always downgrade to previous version. I could not test the pre-release binaries by myself, I will do proper testing once I have time. Thank you for the feedback.

I'm pinging @marelab if they can provide additional thoughts on this.

marelab, 2019-06-23:
Did you add the additional mqtt topics on your mqtt server and enabled mqtt
logging?

bib2, 2019-06-23:
Now it worked to enable MQTT logging and i get logging information on mqtt. I use MQTT.fx so i can see all the messages.

But when i scan a rfid tag, i hear the beep on my reader, but i get no mqtt message from this scan.

bib2, 2019-06-23:
Now i made a downgrade to 1.0.2 and rfid messages with the tag works again. I made an update to 1.3.3 and rfid tags will not be shown as mqtt message.

All other mqtt messages (boot, heartbeat, ...) work, only scanned rfid tag is not working in 1.3.3.

On 1.0.2 i get "access denied" and the uid of my rfid tag. I don´t have any rfid tags saved in my esp, i only want to send the mqtt messages to my server.

bib2, 2019-06-23:
I also want to say another change:

On 1.0.2 my rfid led is green and when i scan a tag i get short blue blinking.

On 1.3.3 the rfid led is blue and when scanning a tag, i get short green blinking.

This behaviour on 1.3.3 is better for me. Where can i change the colors inside esp-rfid?

arretx, 2019-06-25:
Strange.  I'm looking at my interface an I also see 1.3.1, but everything seems to be working so far.  If I scan a tag that has no user, it sends `{"type":"access","time":1561441858,"isKnown":"false","access":"Denied","username":"Unknown","uid":"b9e9c15","door":"door1"}` over to the MQTT broker and the MQTT sensor I created in HA grabs that information and displays it in the States.

Marsx79, 2019-07-04:
Same problem like bib2 with 1.3.3. 
Just boot massage at mqtt.

marelab, 2019-07-04:
That might be the issue because at the GUI you can select MQTT logging or log file on the ESP, if you updated the old config has to be rewritten to have that flag.  If you want MQTT logging ( enable the switch). For details how the MQTT is used (adding additional topics etc.) check this description https://github.com/marelab/esp-rfid until the official Release description.

bib2, 2019-07-04:
I enabled this flag in the GUI (logging only over MQTT), but it was still the old behaviour. Only heartbeats and boot messages, not scan-messages over MQTT.

Marsx79, 2019-07-04:
Me too. I enabled mqtt logging. But it doesn´t  work. I go back to 1.0.3 and all works fine.

theman00011, 2019-07-27:
Can confirm this is a problem on 1.3.1

LEJOUI, 2019-08-12:
I had the same issue. What worked for me was to disable MQTT logging. The topics where the messages are send need to have '/send' after the topic. So for example my topic is 'rfid/send'. I hope this is usefull :)

GeertvanHorrik, 2019-09-02:
Same issue here, even with MQTT logging disabled. "Upgrading" (via the web interface) to 1.0.2 seems to solve the initial issue. I have to reboot quite often since it loses the heartbeats after running for some time.

nerdosity, 2019-09-16:
Same problem here.
I had to downgrade to 1.0.2 to get MQTT tags messages

frenchie71, 2019-09-16:
Can someone who does have the problem please post their setup json (the JSON you see after you click on the yellow bar) on the version that they are having the problem ? So go to the 1.3.x WebUI, select let's say settings-hardware settings, click on "save", click on the yellow bar and cut-past the content ( maybe remove your Wifi password) for us to reproduce ?

drthanwho, 2019-09-28:
Can confirm I have the same problem. When using 1.02 version listening on HASSIO on the mqtt topic (e.g. esprfid), I could receive boot, heartbeat and tag messages.

On 1.3.3 the esprfid topic would only show boot messages. But per @LEJOUI 's suggestion listening on "esprfid/send" I get the tag messages (and no boot messages - they only appear under esprfid).

Haven't received any heartbeats at all though.

Here's the json setup btw:

```
{
  "command": "configfile",
  "network": {
    "bssid": "xxxxx",
    "ssid": "HOME",
    "wmode": 0,
    "hide": 0,
    "pswd": "xxxxxx",
    "offtime": 0,
    "dhcp": 1,
    "ip": "",
    "subnet": "",
    "gateway": "",
    "dns": "",
    "apip": "192.168.4.1",
    "apsubnet": "255.255.255.0"
  },
  "hardware": {
    "readertype": 0,
    "wgd0pin": 4,
    "wgd1pin": 5,
    "sspin": 15,
    "rfidgain": 32,
    "wifipin": 2,
    "rtype": 1,
    "ltype": 0,
    "rpin": 4,
    "rtime": 400,
    "buttonpin": 255,
    "doorstatpin": 255,
    "openlockpin": 255
  },
  "general": {
    "hostnm": "esp-rfid",
    "restart": 18144000,
    "pswd": "admin"
  },
  "mqtt": {
    "enabled": 1,
    "host": "192.168.1.250",
    "port": 1883,
    "topic": "esprfid",
    "user": "mqtt",
    "pswd": "xxxxxx",
    "syncrate": "60",
    "mqttlog": 0
  },
  "ntp": {
    "server": "pool.ntp.org",
    "interval": 30,
    "timezone": 0
  }
}
```

frenchie71, 2019-09-30:
OK I see. Listening on the main topic will not give you all messages. Some of the messages are published on subtopics. So in order to get all messages you would need to subscribe to the main topic plus to the subtopics. With Mosquitto this can be done as follows:

` mosquitto_sub  -h 1.2.3.4 -v -t rfid/#`

In this example my main topic is rfid, the host is 1.2.3.4, the -v switch tells mosquitto to be verbose.

In your case try subscribing to "esprfid/#" and you should see all the subtopics as well. This is not a bug but rather by design.

bib2, 2019-09-30:
Please can you tell me the names of all subtopics, which are possible?

LEJOUI, 2019-09-30:
According to the documentation on https://github.com/marelab/esp-rfid the topics are :
"If you configured over the WebGui for example (TOPIC) = "/rfid" these topic queues can be used:

/rfid
/rfid/sync
/rfid/send
/rfid/accesslist
"

bib2, 2019-09-30:
Thanks a lot, now it is wokring with the new version.

2 small issues:

I installed 1.3.3 but version shows still 1.3.1