omersiar/esp-rfid #281: Door Status Pin - How to use?

arretx, 2019-06-27:
My door status pin, which I've set to GPI05 (there aren't enough pins on my board to use every function in the firmware), seems to respond to touch.

If I touch the pin and hold my finger on it, the board sends open/close/open/close indefinitely until I remove my finger.

How does one go about using the door status pin?

cyberhasse, 2019-06-28:
You can use for example a microswitch or reed sensor..
PIR sensor etc..
Regards.

arretx, 2019-06-28:
Let me be more specific.  If there's a single pin that activates a feature, how to I connect it?  I would presume that one would connect the pin on one side, and the other side of the circuit would be to a GND pin, but I'm suspecting that the GND needs to be something else?  

I tried a reed switch between the status pin and a GND pin and it didn't do anything, but If I touch the pin-side of the connection, MQTT goes bananas with received Open/Close signals.

cyberhasse, 2019-06-28:
Yeah! I had this problem to so i had to change the pin to INPUT_PULLUP in the source code and recompile it..

arretx, 2019-06-28:
Uh oh...now we're talking about doing things I don't know how to do (mainly, converting what's here into some form of editable sketch or something.)  I'm in a Mac/Linux environment and it seems the tools suggested are Windows dependent.

Any suggestions on my next direction, since I can probably figure out the code...

-or- did you perhaps fork the repo with the update?

ver2go, 2019-06-28:
Download the source zip (or clone)

Unzip to a directory.

Download VSCode.
Install Platform.io (from within VSCode)

Open the un-zipped directory in VSCode
Expand src

search for what you want to change.

Click the "build" icon at the bottom of the VScode editor that is part of
Platform.io integration.

If build successful, then click the upload button to the esp.

cyberhasse, 2019-06-28:
I myself is a linux guy and i use platformio to compile the source code..
Install platformio --> download sourcecode --> edit --> compile and upload --> have fun :)

No fork but i changed:
door.esp line 9 and 19 in my source 9 is LOW and 19 is HIGH

arretx, 2019-06-28:
On another note, is there a way to connect directly to my forked repo of esp-rfid through VSCode to open the source code archive, make the change, then commit the change to my repo?

cyberhasse, 2019-06-28:
I have no idea about that.. i only use git clone and keep it locally..

arretx, 2019-06-28:
Thanks...I'm workin' through it now.

omersiar, 2019-08-26:
I'm closing this for now, please feel free to reopen if you want additional help.

closed 2019-08-26 by omersiar