omersiar/esp-rfid #283: Can't add a user?

brumfieldtate, 2019-06-28:
I have a NodeMCU esp8266 connect to my wifi network. I have configured the pins for a RC522 scanner. I double checked to make sure all pins are correct.

When I select users, and scan a rfid tag, nothing happens. I have tried restarting and rebooting.

For some reason it cannot see my RFID tag and register a new user.

I have my hardware settings set as a RC522.

Any ideas??

cyberhasse, 2019-06-28:
Have you set the correct SPI SS gpio pin under hardware settings?

arretx, 2019-06-28:
Keep on it.  Make sure all pins are set correctly and all wiring is correct.  If when it comes down to it you don't get a reading, try a different RC522 scanner.

brumfieldtate, 2019-06-29:
> Have you set the correct SPI SS gpio pin under hardware settings?

Yes I have it sent to GOIO 15. I watched DrZzs setup and figured I’d set one up myself. No clue why tags are not registering. I believe it is the GUI perhaps messing up as sometimes my setting won’t update. Any other ideas? I am on generic bin.

MrAlester, 2019-07-12:
I have the same problem.

Currently using NodeMcu V3 and RC522 (Set as MFRC522). I'm sure the reader is ok because I just test it using a simple code that reads UID.

My pinout is as describe in README file:

D8 -> SDA
D7 -> MOSI
D6 -> MISO
D5 -> SCK
G -> GND
3V -> 3.3V

CONFIG FILE:

```
 {

  "command": "configfile",
  "network": {
    "bssid": "",
    "ssid": "esp-rfid",
    "wmode": 1,
    "hide": 0,
    "pswd": "",
    "offtime": 0,
    "dhcp": 1,
    "ip": "",
    "subnet": "",
    "gateway": "",
    "dns": "",
    "apip": "192.168.4.1",
    "apsubnet": "255.255.255.0"
  },
  "hardware": {
    "readertype": 0,
    "wgd0pin": 4,
    "wgd1pin": 5,
    "sspin": 15,
    "rfidgain": 32,
    "wifipin": 255,
    "rtype": 1,
    "ltype": 0,
    "rpin": 16,
    "rtime": 400,
    "buttonpin": 255,
    "doorstatpin": null,
    "openlockpin": null
  },
  "general": {
    "hostnm": "esp-rfid",
    "restart": 0,
    "pswd": "admin"
  },
  "mqtt": {
    "enabled": 0,
    "host": "",
    "port": 1883,
    "topic": "",
    "user": "",
    "pswd": "",
    "syncrate": 180,
    "mqttlog": 0
  },
  "ntp": {
    "server": "pool.ntp.org",
    "interval": 30,
    "timezone": 0
  }
}

```

MrAlester, 2019-07-14:
Got it working. 

Erased the flash memory, then flashed the DEBUG firmware.

Boot the NodeMcu making sure the RC522 is wired before powering it up, if not, reboot NodeMcu. 

Make sure the serial debugging posts:

`[ INFO ] MFRC522 Version: 0x92 = v2.0`

Connect using web interface and modify only:

**Reader Type**: MFRC522
**SPI SS**: GPIO-15 (D8)
**Antenna Gain**: Max (48 db)

I tried using other pins, but it did not work.

Don't modify Relay Pin, leave it as default **GPIO-4**. Whenever I modified it, it stopped working.

If by any chance the RC522 gets disconnected from the NodeMcu, you must restart it to make it work again.

When it scans a valid card, debug will show:

```
21:30:21.553 -> [ INFO ] PICC's UID: xxxxxx MIFARE 1KB = known PICC
21:30:21.600 -> [ INFO ] User Name: bip have access
21:30:21.600 -> mili : 33155
21:30:21.647 -> activating relay now
21:30:21.953 -> 33555
21:30:21.953 -> 33155
21:30:21.953 -> 400
21:30:21.953 -> 0
21:30:21.953 -> deactivate relay after this
21:30:21.999 -> mili : 33555
```

I used Arduino IDE to debug using COM3.

If you want to make sure your RC522 is working, flash this using Arduino IDE.

```
#include <SPI.h>
#include <MFRC522.h>
#define SS_PIN D8
#define RST_PIN D2

MFRC522 mfrc522(SS_PIN, RST_PIN); // Instance of the class
void setup() {
   Serial.begin(9600);
   SPI.begin();       // Init SPI bus
   mfrc522.PCD_Init(); // Init MFRC522
   Serial.println("RFID reading UID");
}
void loop() {
if ( mfrc522.PICC_IsNewCardPresent())
    {
        if ( mfrc522.PICC_ReadCardSerial())
        {
           Serial.print("Tag UID:");
           for (byte i = 0; i < mfrc522.uid.size; i++) {
                  Serial.print(mfrc522.uid.uidByte[i] < 0x10 ? " 0" : " ");
                  Serial.print(mfrc522.uid.uidByte[i], HEX);
            }
          
            Serial.println();
            mfrc522.PICC_HaltA();
        }
}
}
```

My pinout for both firmwares is this:

```
D8 -> SDA
D7 -> MOSI
D6 -> MISO
D5 -> SCK
G -> GND
3V -> 3.3V
D2 -> Relay
D0 -> RST
```
 
I hope it's helpfull.

brumfieldtate, 2019-07-14:
Thank you so much for all the info.

I am going to try again to see if I get it working, will update you.

brumfieldtate, 2019-07-20:
> Make sure the serial debugging posts:

Is this via Arduino>tools>serial monitor ?

or should I be on Arduion_debug program? This program just gets stuck on "Using proxy DIRECT"

Sorry, inexperienced with debugging esp8266 devices.

Also, i only changed the settings you listed, still no user can be added. I did not change the network/wirless settings either, should i change these so my rfid program can connect to my network as opposedto creating its own access point?

I am trying this on a new nodemcu, and rfid scanner (yes i bought two lol)

MrAlester, 2019-07-20:
First you should select the right COM, it depends on what USB you connect your NodeMcu.

I used [Ctrl] + [Shift] + M  to open the Serial Monitor and set baudrate at 9600.

Check the booting messages displayed at the Serial Monitor and check that it recognizes the rfid scanner.

In my case I used RC522, and if the NodeMcu didn't boot with it connected it wouldn't work, I had to reset it to make it work.

brumfieldtate, 2019-07-21:
`[ INFO ] MFRC522 Version: 0xFF (unknown)`

Thats what I am getting, should it have my version? could it be a wiring issue maybe, not sure?

omersiar, 2019-08-26:
I'm closing this for now, please feel free to reopen if you want additional help.

closed 2019-08-26 by omersiar