omersiar/esp-rfid #284: Change LED-color

bib2, 2019-07-02:
Hi,
with 1.0.2 my LED color in standby is greeen and when i move a rfid-tag to the reader, it flashes blue.

With 1.3.1 my led color in standby is blue and changes to green when i move a rfid-tag near my reader.

Is there a way to set the color for standby and reading by myself?

omersiar, 2019-08-26:
I'm closing this for now, please feel free to reopen if you want additional help.

bib2, 2019-08-26:
Why do you close this? Is there an answer somewhere or have you changed something in the code?

bib2, 2019-08-26:
Maybe you can add this to the next version, so that users can change the led color inside the configuration webif?

Sadly i´m no programmer and i don´t know how to change something in github and build my own source code...

omersiar, 2019-08-26:
What you can do is checking for the output GPIO for your led and find if there any changes were made, then you may edit source code according to your needs. I think there are tools available for git to inspect the difference between versions.

omersiar, 2019-08-26:
Let me know if this case should be remained opened. Kind regards.