omersiar/esp-rfid #285: Scheduled log cleaning

teriall, 2019-07-03:
Hello! After about an year using this awesome project I noticed that after a while I can't see logs at all. For example after 2-3 weeks of using it when I trying to check logs it showing "fetching data" and freezing forever. And I think maybe it's a solution to make option to clear logs every 24 hours for example, or after restart, or every week for example.

omersiar, 2019-07-03:
Thanks again @teriall, now I know the limit. Instead of automized cleaning i will try to make it more robust.

proddy, 2019-07-03:
+1 for this feature. Perhaps purge all logs when remaining SPIFFS space is < a %, or alternatively, implement a rotate so it only keeps the last n entries (like 100).

teriall, 2019-07-03:
> +1 for this feature. Perhaps purge all logs when remaining SPIFFS space is < a %, or alternatively, implement a rotate so it only keeps the last n entries (like 100).

Yes, maybe better to make the option to chose how many entries need be to stored. And make like 100 by default. That would be just perfect!

frenchie71, 2019-09-03:
The logging currently has the following issues:

1. sendEventLog and sendLatestLog go through the whole log file for each page in order to figure out the number of pages each time, hence making n reads of the whole file for n pages
2. bootstrap's footable was not made for very big tables
3. When send_xxx_Log takes too long, the software watchdog triggers and the ESP reboots, thus breaking the Websock connection
4. There is no checking of used SPIFFs space, SPIFFS may get full
5. Once the logfile is too big, there is no way to remediate
6. People using older Keypad implementations might find themselves with a lot of entries in the access log for each  keypress

My suggestions

1. (Quickfix) Increasing Software Watchdog timeout to let's say 4 seconds in the logread functions and setting it back to 1 sec at the end gives a bit more air to breathe
2. Log Files should be rolled over at a given size that corresponds to let's say 10 pages
3. A configurable number of logfiles should be kept, e.g. 4 files - once a 5th is created, the oldest get deleted
4. a global variable NumPages can be assigned on the first logread. The logread function then does not have to read the whole file every time but only for the first page
5. The log viewing function needs to be enhanced to show rolled over logfiles (e.g. lastlog.json, lastlog.1.json etc.)
6. An additional "Maintenance" section could be added to the Web Interface in order to download/split/delete old logfiles

I'd be happy to help out coding some of these; anybody else wants to focus on this ?