omersiar/esp-rfid #286: Last will and testament

cyberhasse, 2019-07-05:
Will you implement "LWT" Last will and testament mqtt feature?
Thanks for your great work.
Regards.

omersiar, 2019-09-14:
#125 #103

Identify all esp-rfids in the network by ChipID
Give user an option to add their topic for LWT

cyberhasse, 2019-11-27:
First of all: I'm no programmer but..
I have LWT success by adding:

		mqttClient.setCleanSession(true);
		mqttClient.setKeepAlive(15);
		mqttClient.setWill("/door/log/lwt", 1, false, "offline");
to config.esp

and adding a function in mqtt.esp
void mqtt_publish_lwt(time_t heartbeat) {
	mqttClient.connect(); // More stable?
  	String stopic (mqttTopic);
	stopic = stopic + "/lwt";
	mqttClient.publish(stopic.c_str(), 0, false, ("online"));

and adding to main.cpp just below nextbeat = (unsigned)now() + interval;

mqtt_publish_lwt(now());

Problem now is that lwt message is hardcoded...

Regards.