omersiar/esp-rfid #288: need some help

zefirpsaryanov, 2019-07-11:
hi again .. :)
i need some help about adding a button for opening the door from inside 
i spent 2 h of reading documentation again and again ... and i can`t understand how to add it 
is it possible atm

teriall, 2019-07-12:
1.3.3 release has button option in Menu > Settings > Hardware settings

zefirpsaryanov, 2019-07-13:
i have downloaded 1.3.3 BUT in esp-rfid-1.3.3.zip all versions are 1.3.1 (generic , debug and forV2Board.bin)

teriall, 2019-07-14:
you need to erase whole flash first, then flash the firmware

zefirpsaryanov, 2019-07-14:
pcb is death .. so i 1st have to revive it .. i have to find where is my ttl usb :)

teriall, 2019-07-14:
There is nothing to die except probably the EPS module. It's cost $2 and you can get in local electronic shops.

zefirpsaryanov, 2019-07-30:
revived .. so i really cannot understood how to add physically button for opening the door 
anybody know or i'll burn it again :)

omersiar, 2019-08-26:
I'm closing this for now, please feel free to reopen if you want additional help.

closed 2019-08-26 by omersiar