omersiar/esp-rfid #292: Question:  Exactly how are you getting the UID of ISO14443-A typed cards?

JohnMcLear, 2019-07-13:
Reading through PN532.cpp I see the logic for supporting Mifare cards and the general UID from NTAG-esque devices but let's say for example you have a Type 4 Card IE SLE77, P5, P6 and you want to get the UID of the SEID.

Currently esp-rfid is getting a UID from these devices which doesn't match the real world UID.

For SLE77, 78, P5, P6 et al the UID is wildly wrong.  Not only the incorrect length but the values are way off.  In these ICs one should primarily target the SEID for a more secure unique identifier.    To achieve this multiple APDUs should be passed until you receive a 9000 response.  I document the most globally common issued GlobalPlatform APDUs on [another project](https://gist.github.com/JohnMcLear/5cef778b3582ee6ebac7db4bb91a5fc8#file-access-js-L144) so feel free to copy/paste them.

The good news is that it is inconsistently wrong so it doesn't break UX but it is possible that the UID value is not going to be unique and unlike the value from an SEID could be easily clonable.  I don't think warrants a specific security issue in itself and that's not my primary concern here.  

My primary query/question is to ensure that the UID esp-rfid is gathering/displaying to a user is accurate and IMHO currently it is not.

So I think some questions need to be answered.

1. Are the UIDs on non-ntag devices correct?
2. Are the UIDs on GlobalPlatform devices correct?
3. If the device is GP should we attempt to query w/ additional APDUs to get the SEID and leverage that over the UID?

I bumped Elmue to see if I'm missing something obvious as he rewrote PN532.cpp.

omersiar, 2019-07-15:
Thank you for your highly valuable feedback,

I always assumed, beside Mifare's tags, getting UID sequence (initial phase) should be same for every type of ISO 14443 PICC (Type A and B).

https://www.iso.org/standard/50942.html

I think we are getting bits and bytes from PICCs correctly but how you represent them -let's say on Web UI- may differ.

I will provide you more details on this when I have time.

Regards.

JohnMcLear, 2019-07-15:
@omersiar Nice one thanks.  I also realized that I was reading PN532 source but actually the test setup was using Weigand.  I need to do some reading about how weigand operates from a protocol level.  I am only doing this as a hobby project over the next 2 weeks so as and when I get time I will try and dip in :)

JohnMcLear, 2019-08-25:
**It appears that non-Mifare enabled Secure Element based cards are not be supported at all.**  The UID will not be visible despite the IC being excited.  I was able to create a solution as per the above APDU commands but I am not confident enough to contribute this back to the project as my solution was in javascript and you guys are leveraging a third party module for querying UID.

So steps to replicate bug:
  1. Get an P5/P6/SLE77/SLE78 et al that doesn't have Mifare enabled.
  1. Present that to a Wiegand reader
  1. No UID will be presented

To get the true SE UID.
  1.  Use the supported GP APDUs: ```80ca9f7f2c 80ca9f7f2d 8002000016 00a4000002``` -- The response will be the UID w/ the trailing 9000 (on success) or 6895 on failure.

While this may seem like a pedantic bug it IMHO can easily brake the UX as lots of issued Smart Cards will not function with the software.  IMHO the bug should be fixed upstream (in the consumed library) so perhaps opening a function there would be more apt..

JohnMcLear, 2019-08-25:
Note I wonder if it's connected to.. https://github.com/monkeyboard/Wiegand-Protocol-Library-for-Arduino/issues/36 -- That's concerning and requires thorough testing..

JohnMcLear, 2019-08-25:
Okay I did more investigation and further reading here: https://github.com/monkeyboard/Wiegand-Protocol-Library-for-Arduino/issues/36

TLDR; The library has two issues.

1. Inconsistent IDs given reader hardware change.  We could hack-patch this by restarting the library upon hardware change.  This is easily tested by this community and validated.  @omersiar if you see the issue on the library it has steps to replicate and you should be able to easily see why we are getting the wrong ID and if reloading the library or some process remedies this *until it's patched properly upstream.

2. Inability to support non-Mifare enabled Smart Cards.  These can be supported but it needs to be really decided on if we want to support this on top of the library or if we hope the library will support them moving forward.

omersiar, 2019-09-10:
Recent change on dev branch by @frenchie71 changed Wiegand Library that we use

https://github.com/beikeland/Wiegand-Protocol-Library-for-Arduino

JohnMcLear, 2019-10-03:
@omersiar I'm no expert and having dig into this but if you are using library A which gives value of X then change to library B that gives value of Y then wont you end up breaking a lot of live installations when they update?

omersiar, 2019-10-04:
The bits and bytes will be same, represtation of these digits will not. You can always convert bits and bytes something more meaningful for the humans ;)

JohnMcLear, 2019-10-04:
Err but isn't the human readable value the one that's stored in the database? :)

omersiar, 2019-10-04:
Yeah and it's not going to change.

closed 2022-02-27 by omersiar