omersiar/esp-rfid #294: Wiegand reader logic levels

teriall, 2019-07-14:
I just noticed that most of wiegand readers require 12V power supply. But as we know ESP8266 works with 3.3V. How does it work together?

teriall, 2019-07-14:
Wikipedia said that wiegand usually has 5V "high" voltage level which should be fatally for the ESP8266. Or is it not that necessarily? I used a reader from aliexpress which has 12V power supply and it works fine over an year.

arretx, 2019-07-15:
On my setup, I run from a 110V supply from the Exit sign power source and step it down to 12V using an LED power converter.  From there, I deliver 12V to a relay, and then to the magnet, and also to a 12V to 5V step down  converter.  This is what powers my NodeMCU which then powers the RFID reader.

beikeland, 2019-07-17:
My reader takes 12-24v ac or dc for the supply, but the wiegand output is 5v. Simple voltage divider 2.2k/3.3k makes it play nice with the 3.3v esp. Just measure the voltage on your signal lines and choose resistors to divide it accordingly to be on the safe side.

omersiar, 2019-08-26:
I'm closing this for now, please feel free to reopen if you want additional help.

closed 2019-08-26 by omersiar