omersiar/esp-rfid #295: esp-rfid connects only to a router with correct mac adress

bib2, 2019-07-20:
Hi, i have some problems:

My esp-rfid connects only to a explicit router with its unique mac adress. Now i have changed the router, everything is working, only the esp rfid doesn´t connect, becuase the new router has a new mac adress.

Can you remove the mac adress from the wlan settings? Every other wlan device connects to a wifi on every router i have in the house, only the esp-rfid connects to the one it was connected with webinterface. So when i move the esp-rfid in my house to a place where another wifi-router is, i don´t have access any more.

arretx, 2019-08-20:
Switch the board back to Access Point mode and connect to it directly, then reconfigure the WIFI to connect back to your WIFI network.

bib2, 2019-08-21:
I did it another way: I sold the old router to a friend, so i called him to bring it back to me and then i was able to connect to the esp again. Then i simply removed the mac adress from the field and it now works with every router.

omersiar, 2019-08-26:
Ok this is a bug, thanks for reporting it.

frenchie71, 2019-09-04:
@omersiar , could you check Pull request #307 again please - I have added a second commit which would fix this issue and issue #251 - thanks ;-)

vrelk, 2019-10-30:
Using the release as of a few days ago, I did have an issue where removing the BSSID caused it to go into AP mode without showing it as such in the web admin. I have since gotten it to act as a client without specifying a BSSID by connecting with a BSSID, then removing it.

matjack1, 2022-02-09:
This should be fixed here: https://github.com/esprfid/esp-rfid/pull/307 in the dev branch for now.

closed 2022-02-09 by matjack1