omersiar/esp-rfid #296: question about timezone?

proddy, 2019-08-02:
The timezone pull-down selection doesn't work for me so I started to look into the code. The timezone selected from the Web UI and stored in the SPIFFS config file is never used in the code, so device time will always be 1 hour + UTC. Is this intentional or is the implementation not finished?

omersiar, 2019-08-02:
Hmm, this sounds completely not intentional, on the other hand I remember debugging a Browser issue, I might be forgot to revert code after testing.

Good finding, I will check it. Thanks

proddy, 2019-08-02:
ok! thought it was me. Thanks and let me know if I can help out with testing or coding.

proddy, 2019-08-16:
Looking at the code again the timezone is taken from the browser (in the JS) so you don't really need to explicitly set a timezone. I think the whole section in the html and config can be removed. Do you agree?

omersiar, 2019-08-18:
Internally, time is being tracked as Unix Time which is then adjusted to User's timezone on Web UI. You need to set timezone in order to list entries in the log correctly.

proddy, 2019-08-18:
Yes, time from NTP is in UTC and when viewed in the event log it's adjusted on the fly to the web browsers's timezone with the code:

```
var comp = new Date();
value = Math.floor(value + ((comp.getTimezoneOffset() * 60) * -1));
```

So you don't need to explicitly set a timezone. In fact it's not even used anywhere in the code so can be removed from the web interface and config file.

omersiar, 2019-08-18:
You are right for the frontend, on firmware level timezone setting is used.

https://github.com/esprfid/esp-rfid/blob/162a7fff662833154172f9defcf382c4be0b0336/src/Ntp.cpp#L23

omersiar, 2019-08-18:
https://github.com/esprfid/esp-rfid/issues/116

proddy, 2019-08-18:
yes it's set in `ntp.cpp` but never used - that's my point. Sorry if I wasn't clear in the beginning. The time is always saved in UTC.

However if you do want to convert the UTC to the correct timezone and handle DST (daylight saving times) in the firmware, I have extended the NTP code to use this [library](https://github.com/JChristensen/Timezone) which works. But now I'm thinking its not really necessary since the only time the Date is ever displayed is in the web interface and JavaScript can do the real-time conversions quite nicely.

omersiar, 2019-08-26:
Oh, now I understand, sorry for the inconvenience. I just need to refactor these functions and libraries.

peteshoard, 2019-09-06:
Time also does not persist during reboot, see images
[image]

matjack1, 2023-10-27:
Fixed by #604

closed 2024-06-11 by matjack1