omersiar/esp-rfid #297: need some help

walliek, 2019-08-05:
hi there,

i have a question, i cant get it working.
i flash the software on the device. (esp2866)
connect to the wifi works, web page works perfectly.
i fill in the wifi settings, hardware setting, general setting. i press safe and reboot.
connect back to my wifi, i can find the device. go to the device ip. fill in password. then i come on the status page but its blance, user page doenst work either and all setting are back the same.  what am i doing rong?
[image]
[image]
[image]

Ralph74, 2019-08-10:
Same problem here. Can flash the esp8266 fine. Device will generate the AP and I can login with the default password however I cannot save ANY changes. It says it is 'Please wait while your ESP restarts', progress bar gets to 100%, says completed. AP stays active. If I reboot ESP, AP still active and any changes have been reverted to default. I tried v 0.9.3, 0.9.4, 1.3.3 Firefox, Chrome, IE. All with the same results. Surely its not the way I'm flashing it (using  NODEMCU FIRMWARE PROGRAMMER)

walliek, 2019-08-10:
I jused also every version, started from 0,7

Tryed also different board, wemos d1 mini 3.1 same issue. Start to think its the rfid reader. Orderd some new ones now.

beroliv, 2019-08-10:
i have the same issue:
My work around as follwow:
 after reboot login
use a menu to make the setting,
 save the new setting and restart..
then next setting you want change. not nice but works so far.

Ralph74, 2019-08-11:
OK. Now trying D1 Mini. I have the console working now with some useful info:
[ WARN ] Failed to open config file
this will be why no settings are saving. Now to work out why?

walliek, 2019-08-11:
found the solution, dont use chrome, safari, firefox, vivaldi. tryed them al. college of mine used microsoft edge and it worked for him. so i reflashed my esp directly, open microsoft edge, setup the everything. en voila it worked... let me know if ut work for u guys

Ralph74, 2019-08-15:
MS Edge did not work for me. @walliek can you please tell exactly how you flashed your device?

arretx, 2019-08-20:
Flashed my device using a Mac with NodeMCU PyFlasher version 4.0.  Flashed the generic.bin from the 1.3.3 patch release.  It's hit or miss it it seems.  Once it takes, it works, but there are other problems with the entire system that I'm trying to figure out.

walliek, 2019-08-21:
@Ralph74 i just use the flash file. What is in the map.

I also order a new RC522. First had 1 from aliexpress. Now i order the one That drzzs linked in is command on youtube. Its working perfect now for me

Ralph74, 2019-08-21:
What the heck is going on now!
[image]
Where did generic.bin go?

walliek, 2019-08-21:
You are on the rong page, https://github.com/esprfid/esp-rfid/releases/tag/1.0.2

omersiar, 2019-08-26:
I'm closing this, please feel free to reopen if you want additional help.

closed 2019-08-26 by omersiar