omersiar/esp-rfid #300: Continuous vs. Momentary

arretx, 2019-08-20:
When a door is magnetically locked, we scan the reader and the lock is powered down.  When the door closes, the lock should already be active again so the door locks when it's closed.

It's suggested that if we use a magentic lock, we would use continuous mode and if we use a striker plate lock we would use momentary mode.

Something doesn't add up regarding this logic.

It seems that without momentary, we would have to have a reader on both sides of the door.  One to deactivate the magnet, and one to activate the magnet, since continuous works like a toggle switch.

In addition, there's no feedback from the system indicating whether or not a lock is active or not.  We can see through MQTT what the last message was and whether or not the door sensor has been tripped, but it seems there's no way to keep MQTT messages in sync with what's actually happening at the door.

Another issue we see is that if we have the lock in Momentary mode, the board boots with the relay turned off, and thus the lock is disengaged.  If we engage the lock and the board loses its connection or reboots for some reason, the lock disengages.

Since the official board's relay closes two circuits, one carrying 12V power (as configured by the pad on the underside) and the other simply connecting the 3rd and 4th screw terminals, we could connect those two pins to the door sensor so the relay would act as a switch that would use the "door open" sensor feature as a "lock active" feature.

By creating a MQTT sensor in Home Assistant, we can run a json template against the MQTT message to know if the door is "Open" or "Closed" but then we lose the ability to sense the state of the actual door.

Is there a way to fire up the board with the relay active by default?

omersiar, 2019-11-06:
The problem with hardware setup is it varies by each use case. What needed is creating real-world objects into something code objects which needs some engineering and coding.

omersiar, 2019-11-06:
> Is there a way to fire up the board with the relay active by default?

Let me check this

closed 2022-03-09 by omersiar