omersiar/esp-rfid #309: Extendet MQTT Features

AlexanderSch90, 2019-09-06:
Hey, i loved to see that now extendet MQTT Support is available!

Can you please tell me how i can use it?
Please give me an example how i can open the door or add and delete an user over MQTT.

Thanks!!

lb-ha, 2019-09-06:
Same for me. I'd like to know the mqtt command to activate the door. Thanks.

marelab, 2019-09-07:
Take a look here  https://github.com/marelab/rfid-door-node-red thats a
complete Node-RED workflow that uses all these MQTT features of the
developer version

AlexanderSch90, 2019-09-07:
@marelab thanks for your work, but i hope this infos will be written down on this readme. I will try your Node-Red Flow. Thanks for sharing...

@lb-ha if i see it right you can test this command: {"cmd":"opendoor","doorip":"192.168.178.123"}

omersiar, 2019-09-10:
Please share with us if you find out any use case with the MQTT functionality.

AlexanderSch90, 2019-09-16:
The MQTT functionality is written down by @marelab here: https://github.com/marelab/esp-rfid

peteshoard, 2019-10-05:
Would it be possible to have the doorbell button generate an MQTT message?

MHOMEASS, 2019-10-23:
Hello, today I'm trying to use the opendoor command fromHA. I'm new to this and I really don't know how to send this command from HASSIO. I prefer not to node red because I want to understand the basics first. I use Mosquitto. Is it possible to use the MQTT command_topic to send the command 'opendoor'?

AlexanderSch90, 2019-10-23:
https://github.com/marelab/esp-rfid/blob/stable/README.md#using-mqtt-topics

MHOMEASS, 2019-10-23:
I already reed thus but I just don´t where to use this in HA :S

{
    cmd:'opendoor',
    doorip:'(The ESP-RFID IP of the door to open as String)'
}

irqnet, 2020-01-26:
> Would it be possible to have the doorbell button generate an MQTT message?

I'm using a piece of code on a standalone ESP to handle that. Would be great if this feature could be implemented.

I can quote the code I'm using later.

matjack1, 2022-11-29:
In the `dev` branch the full MQTT documentation is ready: https://github.com/esprfid/esp-rfid/blob/dev/README-MQTT.md