omersiar/esp-rfid #310: AP mode + RC522 works, Client mode + RC522 not

snauth, 2019-09-06:
Using a Nodemcu v3, esp-rfid 1.0.2 (debug.bin) and a RC522 everything works as expected except when I use the RC522 and Wifi client mode **at the same time**. AP mode and RC522 works, I can add users etc, and Wifi client mode without RC522 connects just fine. Serial monitor shows no errors. Any hints how I can debug this are greatly appreciated. I also tried 1.3.3 but the problem persists. It might be something very simple, this is my first contact with something smaller than a PC.

frenchie71, 2020-01-26:
Hey @snauth , would you mind testing this once more with the 1.3.7 test release please ? Wifi connection has been enhanced in September 2019. Many thanks for you feed-back!

snauth, 2020-01-31:
Working flawlessly with 1.3.7, many thanks!

closed 2020-01-31 by snauth