omersiar/esp-rfid #312: Reboot after each successful access

peteshoard, 2019-09-06:
Device reboots following a successful authentication and lock activation

frenchie71, 2019-09-07:
@peteshoard  , are you sure that the power supply of your board is stable ? I have seen many strange things happen due to voltage variations. If you are running this on a board connected to a USB interface this will definitely not be sufficient

frenchie71, 2019-09-07:
also, which branch are you testing against ? Are you using the dev branch or stable, e.g. a downloaded release ?

peteshoard, 2019-09-07:
Testing against the pre-loaded release 1.0.3.

Currently powering the board off of a 12v LED transformer, think it runs 1a.

I haven't tried measuring the voltage during unlock with the lock connected, only before installation, everything seemed normal. 

Will continue to test and report back, but seems pretty consistent.

arretx, 2019-09-24:
Pete, did you come up with anything on this?  I too am having the same issue and while I haven't tested any voltages yet, I suspect it may be the transformer?  I'm using a Model: 12V-15W Waterproof LED power supply (AC100-265V in / DC12V 1.25A out) and then I believe there's a step-down embedded on the official ESP-RFID relay board.  It reboots almost every time we access it which causes the momentary delay to last much longer than it should.  Also on v1.3.1.

nardev, 2019-09-24:
Guys, if that is the case, it could be board issue.

Please feel free to contact me directly. I'll do my best to fix that.

arretx, 2019-09-24:
Update.  I tested the power to the board.  It comes in at 11.85V.  When the relay is activated and the lock is magnetized, it drops to 11.75.  The 12V feed is powering the relay board and the magnet, as well as a secondary motion control Wemos to trigger the lock via MQTT.  The reboot issue existed prior to adding the Wemos D1 board to the same power feed.

nardev, 2019-09-24:
Anyway, please email me. Lets have a chat to debug it. Maybe i would need to send replacement

peteshoard, 2019-09-24:
I will test mine tomorrow and let you know the results, it's in a
waterproof box outside and it's dark at the moment. The input shares a line
with a light but that is upstream of the transformer, I suspect I will have
similar results but will check and report back ASAP.

Pete

peteshoard, 2019-09-25:
Now Tested:
I see minor differences in the voltage/amperage parameters the board that is functioning correctly/as expected has an input voltage of around 12.40v which drops to just below 12v momentarily as the lock is activated the amperage does not change significantly enough for my meter to measure 0.5a.
The board that is recycling has an input voltage of 11.34v which drops to 11.05v momentarily on lock activation, amperage also does not change 0.5a.

I measured these voltages both at the input terminals on the board, the power terminals on the reader and the relay terminals/ground, apart from minimal differences in the constant voltages between the points tested (likely due to board consumption) I don't see anything significant enough to cause the board to recycle on unlocking, polarity is correct.

I'm, not 100% sure what the screw terminal adjustment is on my boards - does this control output voltage for the relay?

Pete

peteshoard, 2019-10-21:
Any update on this one?

shadow974, 2019-11-11:
same here. running on stable release the esp reboots every now an than after the relais is triggered. Its definetively not an hardware issue as i can trigger the relais with an "blink_demo" endless times without any issue.

closed 2022-11-29 by matjack1