omersiar/esp-rfid #313: MQTT messages do not log device name

peteshoard, 2019-09-06:
When operating 2 esp-rfids with 2 separate doors, there is no way through the interface to pin an identity to the MQTT message to allow identification of a specific door. all messages appear identical.

Current workaround is to include the detail in the User parameters, but this is not ideal for multi door user management.

marelab, 2019-09-07:
In the Dev version the door is identified by the door name and that name is
included in the MQTT message

peteshoard, 2019-09-07:
Is the door name consistent in all messages? Door open/closed and authentication/failed auth?

matjack1, 2026-10-01:
Thanks for raising this, and sorry it sat for so long. Since [2.0.0](https://github.com/esprfid/esp-rfid/releases/tag/v2.0.0) every MQTT message esp-rfid sends includes the device hostname, and access messages also include doorName. Both are set in the web UI (Host Name and Door Name), so with two boards you can tell the doors apart from the payload alone. The full message format is in [README-MQTT.md → Publish Access](https://github.com/esprfid/esp-rfid/blob/stable/README-MQTT.md#publish-access).

Closing this. If something on the current release still can't be traced back to a device, please open a new issue.

closed 2026-10-01 by matjack1