omersiar/esp-rfid #319: Auto Disable Wifi not working in AP mode

adcomrod, 2019-09-26:
I looked first on current open and closed issues but nothing like this found.
I could not make the wifi disabled (using the setting "Auto Disable Wifi") if the node is in access point node.
When set as client is working ok, but as AP the wifi keeps always on, even though the "Auto Disable Wifi" is set to any other than "always on".
I think is useful to be able to disable wifi in standalone mode to add some extra security to the node.
Adrian

omersiar, 2019-09-27:
Thanks for reporting it.

matjack1, 2022-04-05:
Thank you for this report, it makes total sense!

I've fixed it in here: https://github.com/esprfid/esp-rfid/pull/497

I'm closing this for now, but please reopen if necessary.

PR #497 (2022-04-11): Wifi further improvements
I have changed a bit the default timeouts for wifi connections. The issue is that when the connection is retrying, then everything else is blocked, because the testing is happening in the main loop with a call to delay. So changing these timings will reduce the problem overall.

A proper fix would need to be implemented in a completely different way, avoiding the call to delay, which I would like to try avoiding as it would need a major rewrite of the wifi again :)

Then this: https://github.com/esprfid/esp-rfid/compare/dev...wifi-2?expand=1#diff-9d32a6cca07becc2b3812 [... cut]
closed 2022-04-05 by matjack1