omersiar/esp-rfid #320: Reboot with http-request

WAVez86, 2019-09-27:
Hi everyone, real nice project!
I have the problem that the mqtt broker is losing the connection every one till two days.
At the moment my broker sends me a mail and i restart the device with the web-ui per hand.
So I'd like to restart the device automaticly if the mqtt broker is losing the connection. Is there a chance to do this via http-request (e.g. http://IP/api/restart)?
This would help me so much.
Thanks in advance.

adcrafter27, 2022-06-03:
I am also looking for that

matjack1, 2022-10-22:
@WAVez86 @adcrafter27 there's already an issue here: https://github.com/esprfid/esp-rfid/issues/530 to restart the device via MQTT (the currently supported API).

But is your problem occurring with the `dev` branch?

DivanX10, 2022-11-05:
> Hi everyone, real nice project! I have the problem that the mqtt broker is losing the connection every one till two days. At the moment my broker sends me a mail and i restart the device with the [...]

Now it is implemented, you can read the details [here](https://github.com/esprfid/esp-rfid/issues/530#issuecomment-1304606607)

DivanX10, 2022-11-07:
A working reboot command via the MQTT service for this [firmware](https://drive.google.com/drive/folders/15WgpG_04JhWDdBxhBjB9pIWSCzyGV4ll)

```
service: mqtt.publish
data:
  topic: esprfid/cmd
  payload: "{cmd:'reboot',doorip:'192.168.1.48'}"
```

matjack1, 2022-11-07:
I'm going to close this now then

closed 2022-11-07 by matjack1