omersiar/esp-rfid #324: Trying to access Event Log results to a reboot

drthanwho, 2019-10-02:
After logging on to interface and clicking on Event Log it displays the message "Please wait while fetching data..." and seconds later the device reboots itself before being able to show any logs. 

However, Access Log is loading up just fine.

Anyone else had a similar issue?

omersiar, 2019-10-04:
On which version?

drthanwho, 2019-10-04:
Latest dev v1.3.3

rwytrwal, 2019-11-23:
I have the same issue when I have more logs (like couple pages). There is option to send logs just over MQTT and don't save them on the device?

sinclairfr, 2020-02-07:
I also have the same issue and actually could you guys tell us how to simply deactive any logging on the device so that there are less probable freeze or hang because of some reading/writing or even worse memory issues .
Thanks a lot!