omersiar/esp-rfid #325: Access log is empty

clabnet, 2019-10-08:
Using version https://github.com/esprfid/esp-rfid/releases/tag/1.3.3, on NodeMCU Lolin and RFID reader RC522. 
All RFID cards are recognized correctly, but the access log is empty.

bigshit11elf, 2019-10-24:
Same problem here. So far this behaviour is unfixable.

omersiar, 2019-11-06:
There are some great changes since 1.3.3 by @frenchie7. So please, If you are willing to test these changes, I just released some test binaries which can be downloaded here:

https://github.com/esprfid/esp-rfid/releases/tag/1.3.5

Change log can be reviewed here:

https://github.com/esprfid/esp-rfid/blob/dev/CHANGELOG.md

bigshit11elf, 2019-11-17:
Using this firmware the following issues arise:

- Users / Access Log not accessible all the time
- something crashes when RFID is scanned > Users / Access Log not accessible after that
- scanned rfid is not transferred via mqtt

frenchie71, 2020-01-08:
Hi @bigshit11elf , could you please specify a couple of things in order for me to reproduce the issues:

**Users/Access Log not accessible all the time**

What is the symptom ? Does it hang ? How are you trying to access it ? Does the ESP reboot ? Can you define when it is accessible and when not (opening Event Log, then Access Log or opening one log while the other one loads etc.) ?

**Something crashes when RFID is scanned**

Which reader hardware ? All RFIDs ? Only Some ? Always ? Does the ESP reboot ? Do you reconnect to the Webpage after it crashes (because Socket connection will be lost if it reboots, that would be by design)?

**Scanned RFID not transmitted via MQTT**

Which MQTT topics are you listening to ? Have you set the mqtt logging option ?

Many thanks,

sinclairfr, 2020-02-07:
Hello,
 
On version 1.3.1 with :

- official board

- this reader : [https://www.aliexpress.com/item/32849941237.html?spm=a2g0s.9042311.0.0.27424c4dU0wP99](url)

I have the following bugs : 

- hang/freezing and sometimes reboot when trying to change an existing user.

- hang/freezing when reading a unknown rfid badge

- no access event or when there are event they are 1 or 2 days old

- occasional bug/freeze when trying to consult event or access logs

- random hang/freezing after a few hours

When I talk about freezing it is from the webpage but I also ping the board in order to be sure that is is not just another issue and there is no answer to the ping .

matjack1, 2023-01-22:
@clabnet @sinclairfr I'm not sure if you are still interested in this project, but this PR: https://github.com/esprfid/esp-rfid/pull/577 should fix a lot of the stability issues of the web interface.

I'm closing this issue for now, but let me know if you still have issues. Thank you!

closed 2023-01-22 by matjack1

clabnet, 2023-01-24:
@sinclairfr I will try later, Thank's for tips,