omersiar/esp-rfid #326: Cannot build on PlatformIO

clabnet, 2019-10-09:
Using PlatformIO, version 4.0.3 on VSCODE.
I have tried on generic, relayboard and debug environment.
Thre are many build errors, one of the mayor recurrence is :

```
In file included from c:\users\claudio\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\locale_facets_nonio.h:1903:0,
                 from c:\users\claudio\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\locale:41,
                 from c:\users\claudio\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\regex:45,
                 from .pio\libdeps\relayboard\ESP Async WebServer_ID306\src/WebHandlerImpl.h:25,
                 from .pio\libdeps\relayboard\ESP Async WebServer_ID306\src/ESPAsyncWebServer.h:461,
                 from .pio\libdeps\relayboard\ESP Async WebServer_ID306\src\AsyncEventSource.h:29,
                 from .pio\libdeps\relayboard\ESP Async WebServer_ID306\src\AsyncEventSource.cpp:21:
c:\users\claudio\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\locale_facets_nonio.tcc: In member function 'virtual _InIter std::time_get<_CharT, _InIter>::do_get_year(std::time_get<_CharT, _InIter>::iter_type, std::time_get<_CharT, _InIter>::iter_type, std::ios_base&, std::ios_base::iostate&, tm*) const':
c:\users\claudio\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\locale_facets_nonio.tcc:1131:6: error: invalid use of incomplete type 'struct tm'
```
I think is missing one dependency on my platform, but I'm not able to solve it.

omersiar, 2019-10-09:
Yeah, currently I can not build either. Let me find out what's happening on the core side.

clabnet, 2019-10-09:
Here https://pastebin.com/2SsJUSrZ the complete build log.

clabnet, 2019-10-09:
Can you add please .pio on .gitignore file ?

clabnet, 2019-10-09:
And also 
```
[platformio]
default_envs = generic
```
into platformio.ini, because env_default  is deprecated.

omersiar, 2019-10-09:
Please see 5775cd0

omersiar, 2019-10-09:
Everything is now OK according to Travis. Thanks 👍

closed 2019-10-09 by omersiar

clabnet, 2019-10-09:
Sorry, I cannot build, same errors.

omersiar, 2019-10-09:
Please do clean cache and .pio folder

omersiar, 2019-10-09:
And this fix is just on dev branch.

clabnet, 2019-10-09:
Always build errors. :o(
On dev branch: 
- cleared PIO cache folder (C:\Users\Claudio\.platformio\.cache)
- cleared PIO packages folder (C:\Users\Claudio\.platformio\packages)

The last build error log here: 
https://pastebin.com/M41Z3U9R

omersiar, 2019-10-09:
It should not, btw I use pure python version of PlatformIO, no Visual Code extension.