omersiar/esp-rfid #327: Compiling Error

Sanjula-Nipun, 2019-10-10:
anyone know how to fix it

[image]
[image]

[image]

cipiasentini, 2019-10-20:
same error here

bigshit11elf, 2019-10-25:
Had the same errors... created a virtual machine, installed ubuntu, installed "Atom", installed "Python2", installed "PlatformIO" as a package, got it to compile without errors.

Don't know whether this fixes my issues with pre-compiled V1.3.3 (access log empty, wrong version number).

alfredtorch, 2019-11-11:
The same error here. I solved it by adding this to platformio.ini
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
and clean. Still troubleshooting upload (A fatal error occurred: Invalid head of packet (0x80))....

cyberhasse, 2019-11-12:
Try changing upload speed to 115200

matjack1, 2022-03-23:
I'm closing this as I think it's obsolete, but if you still need help let me know!

closed 2022-03-23 by matjack1