omersiar/esp-rfid #330: MQTT send commands

clabnet, 2019-10-25:
Using the 1.2.0 version, I need to get the list of users of rfi-door system.

My flow is : 

1) - publish an MQTT message to _esp-door/accesslist_   with this content: 
{
    cmd:'listusr',
    doorip:'192.168.0.79'
}

2) - subscribe an MQTT message to _esp_door/#

I have not any result. 

The rfi-door system work regularly. It is a bug or I have not understand the commands ? 

Thanks

fabiosoft, 2020-10-01:
Try to swap commands... you should subscribe before publish the message.

matjack1, 2026-10-01:
On 2.x the command was renamed. Send {"cmd":"getuserlist","doorip":"<device IP>"} to TOPIC/cmd, and every user comes back as a separate message on TOPIC/send. Subscribe before you publish so you don't miss the replies (as @fabiosoft said). See [README-MQTT.md → Get all the users](https://github.com/esprfid/esp-rfid/blob/stable/README-MQTT.md#get-all-the-users).

Closing. Please open a new issue if it doesn't work for you on the [latest release](https://github.com/esprfid/esp-rfid/releases/latest).

closed 2026-10-01 by matjack1