omersiar/esp-rfid #331: Auto restart doesn't work

bad-gargoyle, 2019-10-30:
I have problem that my Wemos D1 Pro wont autorestart when I set auto restart option on.

PS Should RST pin (how) be wired?

bad-gargoyle, 2019-11-24:
May we consider the problem confirmed?

runt, 2020-01-30:
I am not really IN, only quickly seeing source main.cpp :

uptime is in seconds ( main.cpp, line 236 = uptime = NTP.getUptimeSec(); )
autoRestartIntervalSeconds is in seconds too

but on line 350 is autoRestartIntervalSeconds multiplied by 1000:

if (autoRestartIntervalSeconds > 0 && uptime > autoRestartIntervalSeconds * 1000)
	{
		writeEvent("INFO", "sys", "System is going to reboot", "");
#ifdef DEBUG
		Serial.println(F("[ WARN ] Auto restarting..."));
#endif
		shouldReboot = true;
	}

if I set in config.json restart=1, then I get autorestart after about 16mins ( 1000s = 16.6667min )

problem confirmed, possibly multiplying in "if" is wrong
sorry for my english

bad-gargoyle, 2020-08-24:
How could I edit/correct (not via GUI) value of autoRestartIntervalSeconds parameter? Is it necessarily to compile new version from source?

PS Could someone pass me link for "know how" for such procedure?

windy54, 2022-06-28:
Install vscode and platform io extension. Download code from GitHub and open in vscode, click on the  “tick” icon to build

matjack1, 2022-11-11:
Fixed here: https://github.com/esprfid/esp-rfid/pull/552

Can anyone please confirm?

matjack1, 2022-11-22:
I'm closing this, in case anyone has problems, please reopen!

closed 2022-11-22 by matjack1

windy54, 2022-11-26:
I have just tried it and it did not re-start. It was supposed to re-set
every 24 hours but has been up for over a day.

I will check the code later on and confirm i have got the latest
development build.

Steve

windy54, 2022-11-26:
Ignore my email, I have jus5 checked the code and have not got the latest, having reviewed the code it will work restart time is in seconds and that is wha5 uptime returns.

Apologies for confusion

Cheers
Steve Gale

matjack1, 2022-11-26:
Thank you for checking, very appreciated!