omersiar/esp-rfid #332: Adding user's role access

clabnet, 2019-10-30:
It is possible to add others user's role access ? 

Seeing the code, the **esprfid.js** file contains the role management, and are expected three access types: _Always, Admin, Disabled._ 
`          {
            "name": "acctype",
            "title": "Access Type",
            "breakpoints": "xs",
            "parser": function(value) {
              if (value === 1) {
                return "Always";
              } else if (value === 99) {
                return "Admin";
              } else if (value === 0) {
                return "Disabled";
              }
              return value;
            },` 

For maintain generic can we use Role1, Role2, Role3, and submit this code to MQTT broker, as is now.  

The problem is not the change of the file, it is hard to minifize the src file and reflash on the board. Thanks.

omersiar, 2019-10-31:
This can only be achieved via code refactoring. Good idea but what exactly roles are good for? other than the currently available ones.

clabnet, 2019-10-31:
Sending different roles via MQTT it is possible to perform different 
actions against the user's role.

omersiar, 2019-11-06:
What do you mean by different actions? Currently esp-rfid only give access or deny it.

clabnet, 2019-11-06:
In a large plus view, If you assign a role to user A, different from user B, and you include this information into MQTT message (as is), when this message will be received from MQTT broker,  you will perform different operations to user A and user B. 
For example, the user A (member of group A-users) will be open the door 1 and 2, and user B (member of group B-users) will be open only the door 1.

omersiar, 2019-11-06:
Oh ok now I understood, You were referring actions from the broker side.

evazzoler, 2020-01-13:
Would be wonderful if a user (tag) would have a OTP role. It works one time, open, then becomes disabled. Please, consider this if you do changes.

evazzoler, 2021-03-19:
Finally I implemented otp serverside, with a good plus. The "life" concept is not "one time" but a number indicating how many times a code can be used. Will set 1 for otp, will become 0 when disabled or leave empty for no limit. But you can set it to 2 or 5 or 157...
Please, consider this for version 2.0. Thanks!