omersiar/esp-rfid #333: ESP8266EX Crash

vrelk, 2019-11-05:
So, I just got a WeMOS D1 Mini clone with an external antenna. It runs off an ESP8266EX, and it seems to crash shortly after booting. I figured I would mention it.

```
18:59:34.817 -> 0⸮~?⸮4⸮Ҷ⸮⸮OCA2⸮⸮
18:59:35.199 -> [ INFO ] ESP RFID v1.3.1
18:59:35.233 -> Flash real id:   00164020
18:59:35.268 -> Flash real size: 4194304
18:59:35.302 -> 
18:59:35.302 -> Flash ide  size: 4194304
18:59:35.302 -> Flash ide speed: 40000000
18:59:35.336 -> Flash ide mode:  DIO
18:59:35.370 -> Flash Chip configuration ok.
18:59:35.404 -> 
18:59:35.648 -> 
18:59:35.648 -> Exception (0):
18:59:35.682 -> epc1=0x4021dbde epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
18:59:35.751 -> 
18:59:35.751 -> >>>stack>>>
18:59:35.751 -> 
18:59:35.751 -> ctx: cont
18:59:35.785 -> sp: 3ffffd10 end: 3fffffc0 offset: 01a0
18:59:35.819 -> 3ffffeb0:  000000f0 3fff1858 0000020f 40220acb  
18:59:35.887 -> 3ffffec0:  0000c000 00001000 3fff1858 40203fb3  
18:59:35.921 -> 3ffffed0:  40220429 4022095c 40220a9c 001fb000  
18:59:35.955 -> 3ffffee0:  00200000 00002000 00002000 00000100  
18:59:36.024 -> 3ffffef0:  00000200 00001fb0 3fff099c 3fff0bfc  
18:59:36.058 -> 3fffff00:  00000000 00000007 3fff1858 402029d9  
18:59:36.127 -> 3fffff10:  00400000 3fff1858 3fff1854 40220740  
18:59:36.161 -> 3fffff20:  400201e9 00400000 3fff099c 40221b29  
18:59:36.229 -> 3fffff30:  00400000 00400000 3fff099c 40245cd3  
18:59:36.263 -> 3fffff40:  feefeffe feefeffe feefeffe feefeffe  
18:59:36.332 -> 3fffff50:  feefeffe feefeffe feefeffe feefeffe  
18:59:36.367 -> 3fffff60:  feefeffe feefeffe feefeffe feefeffe  
18:59:36.435 -> 3fffff70:  feefeffe feefeffe feefeffe feefeffe  
18:59:36.470 -> 3fffff80:  00000002 feefeffe feefeffe feefeffe  
18:59:36.540 -> 3fffff90:  feefeffe feefeffe feefeffe 3fff0ac0  
18:59:36.575 -> 3fffffa0:  3fffdad0 00000000 3fff0a90 4021fc38  
18:59:36.648 -> 3fffffb0:  feefeffe feefeffe 3ffe858c 401013b5  
18:59:36.679 -> <<<stack<<<
18:59:36.713 -> ⸮!⸮⸮⸮⸮!⸮
18:59:36.782 -> [ INFO ] ESP RFID v1.3.1
18:59:36.817 -> Flash real id:   00164020
18:59:36.817 -> Flash real size: 4194304
18:59:36.851 -> 
18:59:36.851 -> Flash ide  size: 4194304
18:59:36.886 -> Flash ide speed: 40000000
18:59:36.921 -> Flash ide mode:  DIO
18:59:36.921 -> Flash Chip configuration ok.
18:59:36.954 -> 
18:59:49.485 -> [ WARN ] Failed to open config file
18:59:49.520 -> [ INFO ] ESP-RFID is running in Fallback AP Mode
18:59:59.855 -> 
18:59:59.855 -> Exception (28):
18:59:59.889 -> epc1=0x4022ab40 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000038 depc=0x00000000
18:59:59.957 -> 
18:59:59.957 -> >>>stack>>>
18:59:59.992 -> 
18:59:59.992 -> ctx: sys
18:59:59.992 -> sp: 3fffed70 end: 3fffffb0 offset: 01a0
19:00:00.026 -> 3fffef10:  3fffdc90 3fff163c 3fff2f0c 4022a9d7  
19:00:00.096 -> 3fffef20:  40252cfa 3fff163c 3fff2f0c 40252d0b  
19:00:00.131 -> 3fffef30:  3fff30ac 3fff309c 00000002 00000000  
19:00:00.166 -> 3fffef40:  4024817f 00000000 3fff2f0c 4024e48b  
19:00:00.235 -> 3fffef50:  40000f49 3fffdab0 3fffdab0 40000f49  
19:00:00.270 -> 3fffef60:  40000e19 40001878 00000002 3fffffc0  
19:00:00.339 -> 3fffef70:  3fffff10 aa55aa55 000000ed 40105244  
19:00:00.374 -> 3fffef80:  4010524a 00000002 3fffffc0 68636174  
19:00:00.443 -> 3fffef90:  4010000d 6966203b 616e656c 223d656d  
19:00:00.477 -> 3fffefa0:  401013f8 3fffef3c 40101399 3ffffe58  
19:00:00.544 -> 3fffefb0:  3fffffc0 00000000 00000000 feefeffe  
[... 288 lines cut ...]
19:00:23.514 -> 3ffff330:  feefeffe feefeffe feefeffe feefeffe  
19:00:23.583 -> 3ffff340:  feefeffe feefeffe feefeffe feefeffe  
19:00:23.618 -> 3ffff350:  feefeffe feefeffe feefeffe feefeffe  
19:00:23.686 -> 3ffff360:  feefeffe feefeffe feefeffe feefeffe  
19:00:23.720 -> 3ffff370:  feefeffe feefeffe feefeffe feefeffe  
19:00:23.789 -> 3ffff380:  feefeffe feefeffe feefeffe feefeffe  
19:00:23.824 -> 3ffff390:  feefeffe feefeffe feefeffe feefeffe  
19:00:23.894 -> 3ffff3a0:  feefeffe feefeffe feefeffe feefeffe  
19:00:23.929 -> 3ffff3b0:  feefeffe feefeffe feefeffe feefeffe  
19:00:23.998 -> 3ffff3c0:  feefeffe feefeffe feefeffe feefeffe  
19:00:24.033 -> 3ffff3d0:  feefeffe feefeffe feefeffe feefeffe  
19:00:24.102 -> 3ffff3e0:  feefeffe feefeffe feefeffe feefeffe  
19:00:24.137 -> {⸮ň!)L⸮<1⸮1⸮⸮
19:00:24.205 -> [ INFO ] ESP RFID v1.3.1
19:00:24.240 -> Flash real id:   00164020
19:00:24.273 -> Flash real size: 4194304
19:00:24.306 -> 
19:00:24.306 -> Flash ide  size: 4194304
19:00:24.341 -> Flash ide speed: 40000000
19:00:24.341 -> Flash ide mode:  DIO
19:00:24.376 -> Flash Chip configuration ok.
19:00:24.409 -> 
19:00:24.409 -> [ WARN ] Failed to open config file
19:00:24.443 -> [ INFO ] ESP-RFID is running in Fallback AP Mode
```

omersiar, 2019-11-05:
Can you please try on different ESP8266 module or development board? At first glance it seems you have a hardware issue.

proddy, 2019-11-05:
A lot of Wemos D1 clones from DO-IT (look at the text on the cap) have faulty WiFi modules which will cause the ESP to reset when the web server starts

vrelk, 2019-11-06:
> 
> 
> A lot of Wemos D1 clones from DO-IT (look at the text on the cap) have faulty WiFi modules which will cause the ESP to reset when the web server starts

Assuming the cap is the long white thing that also kinda looks like a resistor, it is Rainsun. (https://i.imgur.com/xyzLj5W.png)

> 
> 
> Can you please try on different ESP8266 module or development board? At first glance it seems you have a hardware issue.

I previously tried it on a generic NodeMCU and it worked there, but I had an issue where it crashed, but that could have been me feeding 5v into the GPIO ports too. I ended up buying one of yours from the store and figured I would do it the easy way, but I was still hoping someone would know what happened (need a refund or not)

vrelk, 2021-05-20:
Closing this issue as I have since discovered that it is actually the board that is the issue.

closed 2021-05-20 by vrelk