omersiar/esp-rfid #335: Stuck on 0.93

topherslater, 2019-11-10:
0.93 was the first one that i could flash where the settings would stick after a reboot.
for any other version (0.81,1.02,1.33) the settings do not stick. 
My process: upload blink sketch to D1. Copied the command line from the Arduino debug window. Changed the end of the line to point at the generic.bin file. Once flash is done I point my wifi to the new ssid. I a greeted with the status page ( green bars showing memory and such) if i change the anything even just the wifi, to my home SSID, i update it, and I can log in on the home wifi, but the status page is blank and all the settings that I had entered are gone. 

I have uploaded the blank 4mb file.
i have updated from with in the web-app. 0.93-1.02 or 0.93-1.33
same result each time. 
Very cool project. adding the new MQTT features has made me want to pick up again to replace my existing store bought system. 

Using a D1 with pn532. Uploaded with a Mac using terminal. I had tried it before with windows and had the same problems of settings not sticking.

omersiar, 2019-11-11:
Please try different web browser for setting up esp-rfid.

topherslater, 2019-11-21:
Thank you that worked. Just for future people. 
OSX 10.13.6:
Safari 13.0.3 - no 
Chrome  78.39...- no 
Firefox 69.03 -  Yes

closed 2019-11-21 by topherslater