omersiar/esp-rfid #337: 1.3.5 test

beroliv, 2019-11-12:
-Auto-restart is working with the 24h setting
- after loosing the network the device is going back to make his own network (AP Mode)
- after klick the Logs (access and event)the Page is not working properly anymore, first relod the page and login again solve the Problem. I have it on my desk just for testing.
The webui freezes with "**Please wait while fetching data..**."
I recover the user data. I need to do the recover twice till it works and recover the data with each file. (User and System)

beroliv, 2019-11-23:
relay Pin did not work in 1.3.5! After flashing back to 1.3.3 Relay is back working

bad-gargoyle, 2019-11-23:
I dont need relay PIN usage. More important is ability to auto restart after 24h. - So, is it safe for me to use 1.3.5 version?

bad-gargoyle, 2019-11-23:
Sorry, I had to try. ;-) Not only after loosing the network is going to AP mode. After simple restart too. Waiting for final or or least the beta version.

beroliv, 2019-11-24:
i change my wiring so it works with out wifi. 1.3.5 i do no recomment for using

beroliv, 2019-11-24:
what you can try to reenable Wifi is: Set access Typ to admin and switch Wifi off after 10 Minutes

bad-gargoyle, 2019-11-24:
I must do some test before. I'm curious how will MQTT connection will react (will it be still working, how big lags there will be).

beroliv, 2019-11-24:
i try alredy, you have to read the key twice. First read wakup wifi, second read opens the door

bad-gargoyle, 2019-11-24:
Thats not good for me. For now, version 1.3.3 hangs (more or less) once a week. The other thing is that it will not auto reconnect to MQTT when I'm restarting MQTT server/broker. That are main disabilities for me. I have to live withe them for now.

I could always add relay that will reboot my device after each success RFID access. ;-)

beroliv, 2019-11-24:
yes it was my hassel too. i use it no autom with out any wifi and mqtt. opens the door very quick.
Another intressting Projekt but not finish yet.
https://github.com/mdrobisch/roseguarden
https://gitlab.com/roseguarden/roseguarden

i think need a lot of time to finish

bad-gargoyle, 2019-11-24:
In my case all logic and action are implemented and made on central linux server, inside OpenHab. It's better solution, as I could flexibly change RFID scan action for desired OpenHab action depend my actual needs. For examlpe, not only open gate/wicket when scanned, but aditionaly turn on light on front of house before that, if it's dark outside, and aditionally send frame captured from CCTV ans JPG to my telegam acount.

I think I just have to wait for new version.

beroliv, 2019-11-25:
hm but you can try. my beeper is on the relay pin - he made a short beep. Its to short to open the lock but maybe long enough for your needs.

frenchie71, 2019-11-29:
Hi @omersiar , Checking my last pull request I have realized that I had not compiled the websources in the commit. That might actually be the issue as pull request #318 made changes to esprfid.htm and esprfid.js - can you please try again with freshly compiled websources (i.e. run gulp.js and then commit) ? basically the webinterface changed due to the fact that you could now select multiple relays.

matjack1, 2022-03-23:
I'm closing this as it looks obsolete now. But please reopen if you need anything else here.

closed 2022-03-23 by matjack1