omersiar/esp-rfid #339: 1.3.7 Test

beroliv, 2019-11-29:
Hi,
thank you for the new version! Unfortunately there are some problems.

- Network scan are not working
- Cant set admin password
- cant set ssid and Bssid manually

can not safe any settings. After booting still in AP mode, other settings are also lost.

I do before flash first with the blank4mb.bin after i flash the 1.3.7

ingeninge, 2020-01-01:
Hi,
I first installed the stable 1.3.1 and after setting up and testing I found out that MQTT is not working in the stable version, so I downloaded 1.3.7 and compiled it with atom in linux. I did not flash blank4mb but did straight flash 1.3.7 . In WebUi it sais 1.3.1 but MQTT is working also sending a opendoor works. I'm confused and not sure witch version I have now ;) Can you commit that MQTT opendoor only works in versions higher 1.3.1?

Unfortunately when my NodeMCU reboots it activates the door.

beroliv, 2020-01-01:
1.3.3 should work.

ingeninge, 2020-01-02:
Fine, thank you. Tested 1.3.3 same issue. Compiled 1.3.7 with debug and found out I had a retained flag set in node-red. After deleting the retained flag the relais will not be activated on boot :)

Settings can be restored from older versions. Password change without problems also restore of users worked as expected.

frenchie71, 2020-01-08:
Hi @beroliv , I have just tried to reproduce the error situation which you describe. Difference - I am not flashing the image but rather with platformio, i.e.

1. Erased Flash with platformio
2. Built and uploaded debug version
3. Connected to the AP
4. Changed Wifi from Access Point to Client
5. Scanned for networks

Result - All networks showed up.

1. Changed Wifi back to AP
2. Changed Password
3. Rebooted
4. Reconnected to 192.168.4.1

Result: Password has been changed.

So at this time I am unable to reproduce. I am testing on a Wemos D1

Could you advise on your test environment please ?

beroliv, 2020-01-09:
i flash with this Windwos Utility. I use a Nodemcu Board

frenchie71, 2020-01-26:
Grüezi @beroliv , did you manage to get this resolved ? does anyone else have this issue ? I might contact you directly to see if we can troubleshoot if that is OK ?

beroliv, 2020-01-27:
Hallo @frenchie71 this is my spare part. Actually im on a  older version. This device I use for testing and if the produktive one is broken, I have a spare part.

closed 2022-03-23 by matjack1