omersiar/esp-rfid #340: Button pressed - log in the log file and send MQTT

Sansibari, 2019-12-02:
Change Request:
I would like to use the esp-rfid button pin with a push button to open the door from inside. That works as designed.

Unfortunately, pressing the button is not logged in the logs or sent via MQTT.

Please logg in the logs AND also send as MQTT event.
Thanks in advance

BlackbushDarkwood, 2020-01-20:
I also would love to have this feature, too.
So I could get a ring from the doorbell as a MQTT message.

matjack1, 2026-10-01:
This is in since [2.0.0](https://github.com/esprfid/esp-rfid/releases/tag/v2.0.0): a press on the button wired to the Open Lock Pin is written to the access log with user "Button", and published over MQTT as a regular access message with "username": "Button".

For @BlackbushDarkwood's doorbell idea there's now a separate Doorbell Pin. When it rings it writes a "Doorbell ringing" event, and with MQTT for Home Assistant enabled it also publishes on TOPIC/io/doorbell. See [README-MQTT.md → Doorbell](https://github.com/esprfid/esp-rfid/blob/stable/README-MQTT.md#doorbell).

Closing. Feel free to open a new issue if this doesn't cover your setup.

closed 2026-10-01 by matjack1